espressif/esp-idf #3259: handle_dhcp

starlino, 2019-04-05:
Just got this error when station tried  to connect to ESP32 AP. Can't reproduce often, but thought it might be worth mentioning it here for the record and in case someone else has a similar problem.
I am using actually Arduino ESP32, but the error seems to originate in esp-idf

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060330  A0      : 0x8010c509  A1      : 0x3ffb45d0  
A2      : 0x3ffd9bd8  A3      : 0x3ffd9e0c  A4      : 0x3ffda514  A5      : 0x3ffd9200  
A6      : 0x0204a8c0  A7      : 0x0c04a8c0  A8      : 0x8010c3ac  A9      : 0x3ffb4590  
A10     : 0x3ffd9be8  A11     : 0x3ffd9e0c  A12     : 0x3ffb45dc  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb45d0 0x4010c506:0x3ffb4610 0x401193e1:0x3ffb4630 0x4011cded:0x3ffb4670 0x40122022:0x3ffb4690 0x401108d3:0x3ffb46b0 0x4008f2cd:0x3ffb46e0

Decoding 8 results
0x4010c506: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x401193e1: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x4011cded: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40122022: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x401108d3: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 483
0x4008f2cd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)

easyvolts, 2019-06-11:
Hi,
I have some similar problem:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5992
load:0x40078000,len:8800
ho 0 tail 12 room 4
load:0x40080400,len:6380
entry 0x4008072c
[0;32mI (31) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 12:23:23[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 40MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
[0;32mI (67) boot:  1 otadata          OTA data         01 00 0000e000 00002000[0m
[0;32mI (75) boot:  2 app0             OTA app          00 10 00010000 00140000[0m
[0;32mI (82) boot:  3 app1             OTA app          00 11 00150000 00140000[0m
[0;32mI (90) boot:  4 eeprom           Unknown data     01 99 00290000 00001000[0m
[0;32mI (97) boot:  5 spiffs           Unknown data     01 82 00291000 0016f000[0m
[0;32mI (105) boot: End of partition table[0m
[0;32mI (109) boot: No factory image, trying OTA 0[0m
[0;32mI (114) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x394f4 (234740) map[0m
[0;32mI (204) esp_image: segment 1: paddr=0x0004951c vaddr=0x3ffb0000 size=0x030b0 ( 12464) load[0m
[0;32mI (209) esp_image: segment 2: paddr=0x0004c5d4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (211) esp_image: segment 3: paddr=0x0004c9dc vaddr=0x40080400 size=0x03634 ( 13876) load[0m
[0;32mI (225) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x9c8b4 (641204) map[0m
[0;32mI (451) esp_image: segment 5: paddr=0x000ec8d4 vaddr=0x40083a34 size=0x0d4cc ( 54476) load[0m
[0;32mI (472) esp_image: segment 6: paddr=0x000f9da8 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (473) esp_image: segment 7: paddr=0x000f9e14 vaddr=0x50000000 size=0x00d84 (  3460) load[0m
[0;32mI (492) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (492) boot: Disabling RNG early entropy source...[0m
[0;32mI (493) cpu_start: Pro cpu up.[0m
[0;32mI (497) cpu_start: Starting app cpu, entry point is 0x4008117c[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (507) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (514) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (520) heap_init: At 3FFBA220 len 00025DE0 (151 KiB): DRAM[0m
[0;32mI (526) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (533) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (539) heap_init: At 40090F00 len 0000F100 (60 KiB): IRAM[0m
[0;32mI (545) cpu_start: Pro cpu start user code[0m
[0;32mI (138) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m

Free heap in esp32 (bytes): 311924
12[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
Battery voltage(mV): 4024
I (550) wifi: wifi driver task: 3ffc70f8, prio:23, stack:3584, core=0
I (1240) wifi: wifi firmware version: 2af77cc
I (1240) wifi: config NVS flash: enabled
I (1240) wifi: config nano formating: disabled
[0;32mI (1250) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1260) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (1280) wifi: Init dynamic tx buffer num: 32
I (1280) wifi: Init data frame dynamic rx buffer num: 16
I (1280) wifi: Init management frame dynamic rx buffer num: 16
I (1290) wifi: Init management short buffer num: 32
I (1290) wifi: Init static rx buffer size: 1600
I (1290) wifi: Init static rx buffer num: 16
I (1300) wifi: Init dynamic rx buffer num: 16
[0;32mI (1370) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0[0m
I (1370) wifi: mode : sta (24:0a:c4:82:88:84)
I (1370) wifi: mode : softAP (24:0a:c4:82:88:85)
I (1370) wifi: Total power save buffer number: 16
I (1380) wifi: Init max length of beacon: 752/752
I (1380) wifi: Init max length of beacon: 752/752
Free heap after WIFI AP init(bytes): 260100

I (13860) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (13860) wifi: station: 3c:77:e6:b0:cf:d5 join, AID=1, bgn, 40U
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060b30  A0      : 0x80153c0a  A1      : 0x3ffc5790  
A2      : 0x3ffd093c  A3      : 0x00000000  A4      : 0x00000224  A5      : 0x3ffd0b64  
A6      : 0x00000000  A7      : 0x00000224  A8      : 0x801601b1  A9      : 0x3ffc5750  
A10     : 0x3ffd094c  A11     : 0x3ffd0b64  A12     : 0x3ffc5790  A13     : 0x00000044  
A14     : 0x0000002e  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffc5790 0x40153c07:0x3ffc5800 0x40159b2d:0x3ffc5850 0x4015f64e:0x3ffc5880 0x40151fb9:0x3ffc58a0

Rebooting...

Parsed backtrace
0x40153c07: udp_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/cor           e/udp.c:401
0x40159b2d: ip4_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/cor           e/ipv4/ip4.c:740
0x4015f64e: ethernet_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/sr           c/netif/ethernet.c:184
0x40151fb9: tcpip_thread at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/           api/tcpip.c:483

zhangyanjiaoesp, 2019-06-13:
@starlino Please provide the following info:
1. Which IDF version do you use? What is the commit ID of the IDF?
2. How to produce your problem? Just the STA try to connect the AP?
3. Can you provide the *.elf file related to your problem?

Alvin1Zhang, 2019-06-18:
@starlino Thanks for reporting the issue. Would you please help provide the following information, thanks:

1. Which IDF version do you use? What is the commit ID of the IDF?
2. How to produce your problem? Just the STA try to connect the AP?
3. Can you provide the *.elf file related to your problem?

starlino, 2019-06-18:
I apologize but I no longer have the setup, I have upgraded to latest core used in arduino and have no longer seen the error. Feel free to close the ticket and if I run into problem again I will post again with more info. Thank you to all the wonderful people supporting this project !

Alvin1Zhang, 2019-06-19:
@starlino Thanks for the updates.

closed 2019-06-19 by Alvin1Zhang