espressif/esp-idf #3260: a2dp_sink with internal DAC doesn't work

keel-210, 2019-04-05:
## Environment

- Development Kit : ESP32-DevKitC
- Module or chip used : ESP32-WROOM-32D
- IDF version : v3.1.2-99-gcf5dbadf4
- Compiler version : 1.22.0-80-g6c4433a5
- Operating System :  Windows
- Power Supply :  USB

## Problem Description
Hi. I want to modify the sample of a2dp_sink a little and process and output the original audio signal. However, the original audio signal can not be output using the internal DAC.
The code used the original sample and the one corrected in the comment of the issue, but both did not output the correct waveform. And I changed the a2dp_sink config to use the internal DAC as follows:
[image]

[a2dp_sink sample](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink)

### Expected Behavior
I expected that analog waveforms using the internal DAC would be output from GPIO 25 and 26.

### Actual Behavior
I heard noise when I connected to a speaker.
When I looked at the output of the GPIO 25 pin with an oscilloscope, the following waveforms were output.
[image]

According to the following advice, after modifying the code, the GPIO 25 pin outputs the following.
[macaba's modified bt_app_a2d_data_cb](https://github.com/espressif/esp-idf/issues/1614#issuecomment-432858740)
[image]

Console Output
> I (177) I2S: DMA Malloc info, datalen=blocksize=240, dma_buf_count=6
I (177) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, B
CK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14

The above waveform looks more like a digital waveform than an analog waveform.Since these waveforms have a very long period compared to the I2S codec, it can be understood that they output waveforms other than the I2S codec output, but I don't know what waveform they are.

koobest, 2019-04-12:
Hi, @keel-210
Can you provide you configuration of I2S and where is your original audio signal came from? this info  can help us debug this issue.

thanks!!

keel-210, 2019-04-12:
The I2S configuration in main.c is as follows.

> i2s_config_t i2s_config = {
#ifdef CONFIG_A2DP_SINK_OUTPUT_INTERNAL_DAC
        .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
#else
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,                                  // Only TX
#endif
        .sample_rate = 44100,
        .bits_per_sample = 16,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           //2-channels
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count = 6,
        .dma_buf_len = 60,                                                      //
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1                                //Interrupt level 1
    };

And, the original audio signal is sent from my iPhone 8.

keel-210, 2019-04-22:
Analog waveform was output after the version upgrade.

Details are described below.
IDF version : v4.0-dev-402-ga20d02b7f
Other conditions have not changed.
The I2S configuration in main.c is as follows.

> i2s_config_t i2s_config = {
#ifdef CONFIG_A2DP_SINK_OUTPUT_INTERNAL_DAC
        .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
#else
        .mode = I2S_MODE_MASTER | I2S_MODE_TX,                                  // Only TX
#endif
        .sample_rate = 44100,
        .bits_per_sample = 16,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           //2-channels
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,
        .dma_buf_count = 6,
        .dma_buf_len = 60,
        .intr_alloc_flags = 0,                                                  //Default interrupt priority
        .tx_desc_auto_clear = true                                              //Auto clear tx descriptor on underflow
    };

closed 2019-04-22 by keel-210