espressif/esp-idf #3264: undefined reference to `fnmatch'

keelung-yang, 2019-04-06:
on the lastest master

```c
#include <stdio.h>
#include <dirent.h>
#include <fnmatch.h>

int diag_decode_segments(const char * path)
{
    const char * pattern = "[A-Fa-f0-9]+_[A-Fa-f0-9]+.bin";

    DIR * dir = opendir(path);
    struct dirent *entry;
    while ((entry=readdir(dir)) != NULL) {
        int ret = fnmatch(pattern, entry->d_name, FNM_PATHNAME|FNM_PERIOD);
        if(ret == 0) {
            printf("%s\n", entry->d_name);
        }
    }

    return 0;
}

```

```shell
$ make -j4 flash
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/JYANG37/esp-idf/requirements.txt are satisfied.
App "dmodbus" version: v3.4-33-g7f6da69-dirty
CC build/main/diag.o
AR build/main/libmain.a

Generating libmain.a.sections_info
Generating esp32.project.ld
LD build/dmodbus.elf
C:/msys32/home/JYANG37/bootloader/esp32/build/main\libmain.a(diag.o):(.literal.diag_decode_segments+0x4): undefined reference to `fnmatch'
C:/msys32/home/JYANG37/bootloader/esp32/build/main\libmain.a(diag.o): In function `diag_decode_segments':
C:/msys32/home/JYANG37/bootloader/esp32/main/diag.c:48: undefined reference to `fnmatch'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/msys32/home/JYANG37/esp-idf/make/project.mk:517: /home/JYANG37/bootloader/esp32/build/dmodbus.elf] Error 1
```

igrr, 2019-04-08:
Hi @keelung-yang, indeed the functions implemented in [`posix` directory of newlib](https://github.com/espressif/newlib-esp32/tree/esp32-2018r1_newlib-2_2_0/newlib/libc/posix) are not present in our newlib builds. Among these are all regular expression functions, wordexp, fnmatch, process creation functions, and directory related functions. Some of the directory related functions have been implemented in IDF by the VFS layer. In one of the future toolchain updates we will include `posix` directory into our newlib builds, so that the rest of the functions are available.

hansmbakker, 2019-09-01:
I would like to have this functionality as well, +1 for this request!