espressif/esp-idf #3265: Both build systems stop working if time is incorrect

huming2207, 2019-04-06:
## Environment

- Development Kit:      n/a
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-266-ge7f85f198
- Build System:  CMake and Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Manjaro Linux v18, latest update
- Power Supply:        n/a

## Problem Description

If the ESP-IDF is downloaded later than the project is created, then both of the build systems will fail.

This could happen when someone travelling on air between two different timezones, or just a wrong time was set to the host system previously (and get corrected later). 

### Expected Behavior

Projects should be built successfully, even the file timestamps in IDF is later than the project.

### Actual Behavior

CMake build system: it will repeat forever:

```
idf.py menuconfig
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /home/hu/esp/esp-idf/requirements.txt are satisfied.
Running cmake in directory /home/hu/Projects/remote_log/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /home/hu/Projects/remote_log"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.21.0") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: xtensa esp_common esp_rom soc log heap app_trace freertos vfs newlib esp_ringbuf driver mbedtls wpa_supplicant efuse bootloader_support partition_table app_update spi_flash nvs_flash esp_wifi ethernet lwip tcpip_adapter esp_event pthread smartconfig_ack espcoredump esp32 cxx asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: /home/hu/esp/esp-idf/components/xtensa;/home/hu/esp/esp-idf/components/esp_common;/home/hu/esp/esp-idf/components/esp_rom;/home/hu/esp/esp-idf/components/soc;/home/hu/esp/esp-idf/components/log;/home/hu/esp/esp-idf/components/heap;/home/hu/esp/esp-idf/components/app_trace;/home/hu/esp/esp-idf/components/freertos;/home/hu/esp/esp-idf/components/vfs;/home/hu/esp/esp-idf/components/newlib;/home/hu/esp/esp-idf/components/esp_ringbuf;/home/hu/esp/esp-idf/components/driver;/home/hu/esp/esp-idf/components/mbedtls;/home/hu/esp/esp-idf/components/wpa_supplicant;/home/hu/esp/esp-idf/components/efuse;/home/hu/esp/esp-idf/components/bootloader_support;/home/hu/esp/esp-idf/components/partition_table;/home/hu/esp/esp-idf/components/app_update;/home/hu/esp/esp-idf/components/spi_flash;/home/hu/esp/esp-idf/components/nvs_flash;/home/hu/esp/esp-idf/components/esp_wifi;/home/hu/esp/esp-idf/components/ethernet;/home/hu/esp/esp-idf/components/lwip;/home/hu/esp/esp-idf/components/tcpip_adapter;/home/hu/esp/esp-idf/components/esp_event;/home/hu/esp/esp-idf/components/pthread;/home/hu/esp/esp-idf/components/smartconfig_ack;/home/hu/esp/esp-idf/components/espcoredump;/home/hu/esp/esp-idf/components/esp32;/home/hu/esp/esp-idf/components/cxx;/home/hu/esp/esp-idf/components/asio;/home/hu/esp/esp-idf/components/jsmn;/home/hu/esp/esp-idf/components/aws_iot;/home/hu/esp/esp-idf/components/bootloader;/home/hu/esp/esp-idf/components/bt;/home/hu/esp/esp-idf/components/coap;/home/hu/esp/esp-idf/components/console;/home/hu/esp/esp-idf/components/nghttp;/home/hu/esp/esp-idf/components/esp-tls;/home/hu/esp/esp-idf/components/esp_adc_cal;/home/hu/esp/esp-idf/components/tcp_transport;/home/hu/esp/esp-idf/components/esp_http_client;/home/hu/esp/esp-idf/components/esp_http_server;/home/hu/esp/esp-idf/components/esp_https_ota;/home/hu/esp/esp-idf/components/openssl;/home/hu/esp/esp-idf/components/esp_https_server;/home/hu/esp/esp-idf/components/esptool_py;/home/hu/esp/esp-idf/components/expat;/home/hu/esp/esp-idf/components/wear_levelling;/home/hu/esp/esp-idf/components/sdmmc;/home/hu/esp/esp-idf/components/fatfs;/home/hu/esp/esp-idf/components/freemodbus;/home/hu/esp/esp-idf/components/idf_test;/home/hu/esp/esp-idf/components/json;/home/hu/esp/esp-idf/components/libsodium;/home/hu/esp/esp-idf/components/mdns;/home/hu/esp/esp-idf/components/mqtt;/home/hu/esp/esp-idf/components/protobuf-c;/home/hu/esp/esp-idf/components/protocomm;/home/hu/esp/esp-idf/components/spiffs;/home/hu/esp/esp-idf/components/ulp;/home/hu/esp/esp-idf/components/unity;/home/hu/esp/esp-idf/components/wifi_provisioning;/home/hu/Projects/remote_log/main
-- IDF_VER: v4.0-dev-266-ge7f85f198
-- Project version: 126587f
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Found PythonInterp: /usr/bin/python (found version "3.7.3") 
-- Found Perl: /usr/bin/perl (found version "5.28.1") 
-- Adding linker script /home/hu/Projects/remote_log/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Building empty aws_iot component due to configuration
-- Component libraries: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hu/Projects/remote_log/build
Running ninja in directory /home/hu/Projects/remote_log/build
Executing "ninja menuconfig"...
[... 190 lines cut ...]
make[1]: Warning: File '/home/hu/esp/esp-idf/components/xtensa/component.mk' has modification time 28619 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/wpa_supplicant/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/wifi_provisioning/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/wear_levelling/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/vfs/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/unity/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/ulp/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/tcpip_adapter/component.mk' has modification time 28618 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Warning: File '/home/hu/esp/esp-idf/components/tcp_transport/component.mk' has modification time 28618 s in the future
```

### Steps to repropduce

1. Set the system timezone or time to a wrong one, to the future
2. Download ESP-IDF as instructed
3. Set back the system time/timezone to the correct, current one
4. Grab a project and compile

projectgus, 2019-04-09:
@huming2207 Thanks for reporting this, we'll fix it ASAP.

Silvesterrr, 2022-11-05:
I've reinstalled esp-idf and saved again every file in my project and it fixed the problem...

zikalino, 2023-11-29:
Closing this ticket as it's pretty old already.
In addition this problem happend in a very rare condition of changing timezones and it really can't be fixed in ESP IDF.

closed 2023-11-29 by zikalino