espressif/esp-idf #3267: Channel State Information (CSI) sig_mode

kraoa, 2019-04-06:
Hi,
In the CSI call back function, I'm getting frames having sig_mode = 0. This sig_mode = 0 corresponds to 802.11bg (per esp documentation). However I read elsewhere CSI was included in the 802.11 standard only starting at 802.11n which is newer than 802.11bg. So can someone explain why this is happening. Is this is a bug?
Also the CSI results using sig_mode = 1 are quite noisy i.e. the amplitude and phase responses across sub-carriers are not slowly varying but change drastically from one sub-carrier to the next.
I used the toolbox
https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-

and am getting similar results as shown on the above link.
If you compare this with other CSI results acquired using non-esp32 chips , then you get nice results as shown in Figure 3 in
http://www.cs.wm.edu/~yma/files/WiFiSensing_YongsenMa_authorversion.pdf

There are many other papers like the one I showed above in which signal changes slowly across sub carriers.
Karthik

TimXia, 2020-05-29:
@kraoa CSI is carried by OFDM frame. 11g also uses OFDM. So there is no problem with sig_mode=0. 
For the second question, we will take a look at it.

zhangyanjiaoesp, 2020-10-29:
@kraoa 
Have you seen the description for the Figure 3 ? I think that is why it is different from the result in https://github.com/jonathanmuller/ESP32-gather-channel-state-information-CSI-.
[image]

closed 2020-12-09 by Alvin1Zhang