espressif/esp-idf #3269: BLE advertising Bug

Andy001847, 2019-04-07:
When the esp32 as client and server, use client to receive ble device data, then as a server to advertising some data. It would occur the mistash and do not work any more. The information is "ASSERT_ERR(bt_intcntl_finetgtintmsk_getf() == 1), in ea.c at line 497"

Weijian-Espressif, 2019-04-08:
@Andy001847, What is the commit id of ESP-IDF ? Could you provide the test log? Thanks.

Alvin1Zhang, 2019-04-12:
@Andy001847 Hi, would you please help provide commit id of ESP-IDF as well as the test log? Thanks.

Alvin1Zhang, 2019-05-16:
@Andy001847 Thanks for reporting the issue. Will close this for now due to lack of details. Feel free to reopen and please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md). Thanks.

closed 2019-05-16 by Alvin1Zhang