espressif/esp-idf #3271: Failed to find Xtensa toolchain

ghmeyes, 2019-04-07:
hello:
    I follow the steps of Standard Setup of Toolchain for Linux
Install Prerequisites,but failed.
   
   my system is  ESP8266_lubuntu_20141021.ova

   esp8266@esp8266-VirtualBox:~/esp/project_template$ printenv PATH
/opt/xtensa-lx106-elf/bin:/home/esp8266/esp/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/esp8266/esp/xtensa-lx106-elf/bin:/home/esp8266/esp/xtensa-lx106-elf/bin
esp8266@esp8266-VirtualBox:~/esp/project_template$ make
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
LD build/bootloader/bootloader.elf
/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
make[1]: *** [/home/esp8266/esp/project_template/build/bootloader/bootloader.elf] Error 1
make: *** [/home/esp8266/esp/project_template/build/bootloader/bootloader.bin] Error 2
esp8266@esp8266-VirtualBox:~/esp/project_template$

igrr, 2019-04-08:
Hi @ghmeyes, when you were opening this issue, you have been presented with the issue template: https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md. Please update the issue description, providing the information requested therein.

Also please mention the following:
* Whether your intention is to build a project for ESP8266 or the ESP32?
* Your sdkconfig file
* Output of `echo $PATH`

Thanks.

negativekelvin, 2019-04-08:
Did you complete Toolchain Setup section?
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html

ghmeyes, 2019-04-08:
thank you @igrr!
     This is my first time show my issue in GitHub,I am so excited when I receive your answers,thank you very much!
     How do I build a project for ESP8266.
    $PATH:
esp8266@esp8266-VirtualBox:~$ echo $PATH
/opt/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/esp8266/esp/xtensa-lx106-elf/bin

  I don't know what is the sdkconfig file. I use this "git clone --recursive https://github.com/espressif/ESP8266_RTOS_SDK.git" to get SDK.
 
**xtensa-lx106-elf-gcc --version**
esp8266@esp8266-VirtualBox:~$ xtensa-lx106-elf-gcc --version 
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Operating System: oracle VM VitualBox Linux(ESP8266_lubuntu_20141021.ova)
                                                                                   thank you!

ghmeyes, 2019-04-08:
thank you @negativekelv!
I follow this Toolchain Setup section:
https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html#get-esp8266-rtos-sdk

But I can't run below command.
python2.7 -m pip install --user -r $IDF_PATH/requirements.txt

igrr, 2019-04-08:
@ghmeyes since you are working with the ESP8266 RTOS SDK, please open future issues in https://github.com/espressif/ESP8266_RTOS_SDK/issues. I'm going to move this one there as well.