espressif/esp-idf #3272: Connect to AP without authentication

Vakaras1, 2019-04-07:
## Environment
- Module or chip used:  [ESP32-WROOM-32]
- IDF version: v3.1.3
- Build System:         [Make]
- Compiler version:  1.22.0-80-g6c4433a5
- Operating System:     [Windows]
- Power Supply:         [USB]
## Problem Description
WROOM-32 Wifi can not connect to open access point (without password protection).
After startup, I receive the SYSTEM_EVENT_STA_START event. After that continuously (every few seconds) appears SYSTEM_EVENT_STA_DISCONNECTED event with the reason code 201
If I setup my router (AP) to have a password - wroom-32 connects to it without problems, it receives IP.
### Expected Behavior
Expecting to receive a SYSTEM_EVENT_STA_GOT_IP event.
### Actual Behavior
Receive SYSTEM_EVENT_STA_DISCONNECTED instead of SYSTEM_EVENT_STA_DISCONNECTED 
Here is the printout:
[image]

### Steps to repropoduce

### Code to reproduce this issue

```cpp
static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        xEventGroupSetBits(wifi_event_group, WIFI_CONNECTED_BIT);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
    	ESP_LOGI(TAG, "Disconnecting from %s. Reason: %d\n", event->event_info.disconnected.ssid, event->event_info.disconnected.reason);
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, WIFI_CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

void app_main()
{	esp_err_t ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
	{
	        ESP_ERROR_CHECK(nvs_flash_erase());
	        ret = nvs_flash_init();
	}
	ESP_ERROR_CHECK( ret );
	wifi_event_group = xEventGroupCreate();
	tcpip_adapter_init();
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        wifi_config_t sta_config = {
        .sta = {
            .ssid = "Mano Tvirtove",
   	    .password = ""
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config)); 
    ESP_ERROR_CHECK(esp_wifi_start());
    while (1) {	vTaskDelay(1000);  }
}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs
## Other items if possible

liuzfesp, 2019-04-23:
Hi @Vakaras1, I suspect you connected the open AP with configured password.

Vakaras1, 2019-04-23:
Re: [espressif/esp-idf] Connect to AP without authentication (IDFGH-928) (#3272)

Hello liuzfesp,

    Yes. Not sure, if it was the correct way, but I did the following:
  - made AP scan;
  - Wrote MAC address of my AP after scan;
  - Tryed to connect to my AP.

  This sequency helped. Probably a MAC address in configuration helped, but didn't researches. Just happy, the solution works.

Regards,
Vakaras

Hi @Vakaras1, I suspect you connected the open AP with configured password.
—

Best regards,
 Vakaras                            mailto:vakaras@e-kaunas.com

Alvin1Zhang, 2019-11-07:
@Vakaras1 Thanks for reporting, feel free to reopen if the issue still exists. Thanks.

closed 2019-11-07 by Alvin1Zhang