espressif/esp-idf #3273: Weird function output parameter

keelung-yang, 2019-04-08:
on the lastest master.
[image]

[image]

[image]

igrr, 2019-04-08:
Hi @keelung-yang, indeed based on the screenshots this looks odd. However without you providing  a self-sufficient code snippet to reproduce this, I don't think we can start investigating this behavior.

Also, while this might not be immediately relevant, it seems that there are some buffer overflows in the code displayed? For example, `sprintf(req, "3101FF01%04X"` will write 13 bytes into a buffer of size 8. The other sprintf seems to be writing 24 bytes into a buffer of size 16.

keelung-yang, 2019-04-08:
@igrr Exactly! It's my mistake and shouldn't be here. Sorry for that.

closed 2019-04-08 by keelung-yang