espressif/esp-idf #3274: Can't establish Serial Connection

GeorgeFlorian, 2019-04-08:
OS: Linux Mint 19.1
Board: ESP32 Wrover-B DevKitV4

## Issue: 
Can't establish Serial Connection

## Summary:

Hello ! I've been using Arduino IDE and Platformio IDE for the past 2 months and today I tried installing ESP-IDF because I believe it can help me with a project.

The problem is that I can't establish a Serial Connection with the ESP. It works fine with Arduino IDE, I can upload code, etc, but when using Putty I have no output.

I followed this guide: https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-started-connect

And then I ended up here: https://docs.espressif.com/projects/esp-idf/en/stable/get-started/establish-serial-connection.html

I can't move past this because I have no output on the Serial Terminal !

`/dev/ttyUSB0` is my port !

I've used `sudo usermod -a -G dialout $USER`, then restarted the PC.

I've opened Putty and used the following settings: https://i.postimg.cc/26qc36FL/devtty.png

I've opened a terminal and typed: `screen /dev/ttyUSB0` and I have one of the following outputs:

1. `[screen is terminating]`
2. No output at all

What can I do ? Thank you !

koobest, 2019-04-10:
Hi, @GeorgeFlorian 

Is it possible that you select a wrong port?  maybe should ttyUSB1 or ttyUSB2  or you turned on the flow control of the Putty.

GeorgeFlorian, 2019-04-10:
> Hi, @GeorgeFlorian
> 
> Is it possible that you select a wrong port? maybe should ttyUSB1 or ttyUSB2 or you turned on the flow control of the Putty.

I know for sure that ttyUSB0 is the port. But, I've also tried other ports and none worked.
I've never used Putty, so I don't know what a flow control is or what it's default setting was.

koobest, 2019-04-11:
@GeorgeFlorian 
 I don't know if you are using a virtual machine. You can re-test it on Arduino IDE Platformio to make sure the board is OK,  and then,  you can try another serial port assistant. 

thanks !!

me-no-dev, 2019-04-11:
I just had a discussion with an Arduino user yesterday and he also experienced issues using Putty and ESP32. The problem is that putty does not handle the RTS and DTR lines properly and is putting the ESP into reset state.

In the Putty Settings > Connection > Serial you will see a drop down for "Flow control". Try different settings there and see which one will work

koobest, 2019-04-11:
Ok, so, this is the flow control issue :-) .  If  someone encounter the same problem we will know how to solve this problem.

thanks !!

GeorgeFlorian, 2019-04-11:
> I just had a discussion with an Arduino user yesterday and he also experienced issues using Putty and ESP32. The problem is that putty does not handle the RTS and DTR lines properly and is putting [...]

Thanks !
Will try it out tomorrow.

Alvin1Zhang, 2019-06-11:
@GeorgeFlorian Thanks for reporting the issue, would you help share if any updates for the issue? Thanks.

GeorgeFlorian, 2019-06-11:
> @GeorgeFlorian Thanks for reporting the issue, would you help share if any updates for the issue? Thanks.

I wasn't able to make it work !
I threw the ESP away and got a new one.
I really think it was broken.

closed 2019-06-11 by GeorgeFlorian