espressif/esp-idf #3275: ESP32 in Access Point Mode: clients disconnected shortly

thecashit, 2019-04-08:
Hello there!

ESP-IDF v3.2-rc
While ESP-IDF in Access Point mode i'm unable to connect from phone or laptop and establish reliable connection.

Very shortly i have this message in monitor and client disconnecting (it appears from 10 to 60 seconds):
dhcps: send_offer>>udp_sendto result 0

Also i tried to disable AMDPU RX (in sdkconfig and in wifi_config too) with no luck.
DHCP server settings is default.

How can i solve this?
On 3.1 there no such problems.

Thnx.

zhangyanjiaoesp, 2019-06-19:
@thecashit Would you please help provide the following information, thanks:
1. How to reproduce this problem?
2. If you just use an iphone or a laptop to connect the ESP32 softAP, please provide the model of the iphone or the laptop.
3. If possible, please provide more log about your issue.

thecashit, 2019-10-15:
@Alvin1Zhang 
@zhangyanjiaoesp 

Seems like this issue is not actual right now.
I solved this by enabling WiFi AMPDU TX / RX settings as i read in internet forums.
Don't know how but it works now.

closed 2019-10-15 by thecashit

zhangyanjiaoesp, 2019-11-06:
@thecashit  Although you've solved the problem, I still don't know  what your problem is.
 At the very beginning, you said `disable AMDPU RX with no luck`, but in the end, you said you `solved this by enabling WiFi AMPDU TX / RX settings`, this is very confusing.
It will be appreciate it if you can share the ins and outs of the problem, thanks.

thecashit, 2019-11-11:
@zhangyanjiaoesp 
Hello!

The fact is that about 6 months have passed and during this time, several versions of ESP-IDF have changed, just like the settings were changed several times. Unfortunately, now I cannot reproduce and explain step by step what exactly the problem was, as well as explain the exact way to solve it from our side.

I propose to close this ticket, the next time when creating a new one I will try to describe as much as possible what is happening, and also log for myself the solution paths if they are found so that such discrepancies do not occur.

Thank you for participating, but I consider it irrelevant at the moment.

zhangyanjiaoesp, 2019-11-11:
@thecashit Thanks for your reply