espressif/esp-idf #3276: changed "UART_RX_SIZE" have no effect

GitOfLance, 2019-04-08:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
In [ESP32 Technical Reference Manual(PDF)](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) 13.3.3  Figure 81, I can expand the memory of  "Rx_FIFO" by change “UARTn_RX_SIZE”.But I can't find "UARTn_RX_SIZE" or "UART0_RX_SIZE" in the project, I can only find "UART_RX_SIZE" in "esp-idf/components/soc/esp32/include/soc/uart_reg.h ", the code is as follows:
```
/* UART_RX_SIZE : R/W ;bitpos:[6:3] ;default: 4'h1 ; */
/*description: This register is used to configure the amount of mem allocated
 to receiver's fifo. the default byte num is 128.*/
#define UART_RX_SIZE  0x0000000F
#define UART_RX_SIZE_M  ((UART_RX_SIZE_V)<<(UART_RX_SIZE_S))
#define UART_RX_SIZE_V  0xF
#define UART_RX_SIZE_S  3
```
At first I wanted to expand the memory of "Rx_FIFO", which is bigger than the default of 128 Bytes, but I found that this parameter is only 4 bits, the maximum is 0x0F, so I want to try it can be smaller, and finally found that no matter how I modify this value, interrupt event The maximum number of bytes when “UART_DATA” is triggered is 120 Bytes.

The code I am using is an example provided  [peripherals/uart_events](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_events) ，just adding some printf informations.
### Debug Logs
```
I (120059) uart_comm: uart[1] event:
I (120059) uart_comm: [UART DATA]: 120
-~~-
I (120069) uart_comm: uart[1] event:
I (120069) uart_comm: [UART DATA]: 120
-~~-
I (120079) uart_comm: uart[1] event:
I (120079) uart_comm: [UART DATA]: 120
-~~-
I (120089) uart_comm: sending data..
I (120089) uart_comm: send 0 data
I (120089) uart_comm: uart[1] event:
I (120089) uart_comm: [UART DATA]: 120
-~~-
I (120099) uart_comm: uart[1] event:
I (120099) uart_comm: [UART DATA]: 38
-~~-
```

If I want to modify the Rx_FIFO, how to operate, is the maximum memory size 128 bytes?

negativekelvin, 2019-04-08:
Do you really need to change the fifo size?  When the fifo fills up to the RX threshold (120), the interrupt will copy the data to the ring buffer which you can make as big as you want.  If you are losing data first try lowering the rx threshold.

GitOfLance, 2019-04-08:
Whether modifying or not modifying the fifo size, I can achieve the effect I want by programming, but I want to know why the operation fails according to the instructions. Is this method invalid? Or am I doing something wrong?

negativekelvin, 2019-04-08:
If you do not change the threshold the data will still be copied out of the fifo at 120 even if the fifo is bigger.

GitOfLance, 2019-04-08:
I see, I think I know what to do, thank you very much.

So, for the user, it is not necessary to modify the fifo size?

negativekelvin, 2019-04-08:
It should not be necessary for vast majority of cases.

GitOfLance, 2019-04-08:
Ok,  I found that there is a macro `UART_FIFO_LEN` in `driver/uart.h` that sets the size of the hardware FIFO buffer. This value is the upper limit of the threshold  [uart_enable_tx_intr](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/api-reference/peripherals/uart.html#_CPPv419uart_enable_tx_intr11uart_port_tii)   , and up to 128 bytes, because if I increase this parameter, an error will occur. I used UART1 by GPIO18,GPIO19
```
E (360) uart: uart_driver_install(1330): uart rx buffer length error(>128)
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40088030
```
So, I think the FIFO buffer size can only be up to 128 bytes in length, and it can't be bigger. Is that true?

negativekelvin, 2019-04-08:
Larger fifo is not supported by the driver but per TRM it is technically possible.

GitOfLance, 2019-04-08:
Ah, what is TRM? Can you put it simply?

negativekelvin, 2019-04-08:
ESP32 Technical Reference Manual

GitOfLance, 2019-04-08:
Oh, I understand, thank you for your patience and professional answers, thank you very much!!!

Alvin1Zhang, 2019-06-11:
@negativekelvin Thanks for the notes. @GitOfLance Thanks for reporting the issue, would you help share whether your issue has been resolved? Thanks.

GitOfLance, 2019-06-18:
@Alvin1Zhang 
This problem has been solved, each time the data is received, it is put into the linked list node, and another task is used to process the splicing and processing of the data.

Alvin1Zhang, 2019-06-18:
@GitOfLance Thanks for the updates.

closed 2019-06-18 by Alvin1Zhang