espressif/esp-idf #3277: stat() function on SPIFFS is too slow

no1seman, 2019-04-08:
- Development Kit:      ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER
- IDF version:  v4.0-dev-266-ge7f85f198
- Build System: CMake
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description
I my program logic I have to check file existance during config checks processing (see code in Code to reproduce this issue section). When I call is_file_exist() function during boot (before setting up Network, BLE, MQTT, etc. I've got 1,333 seconds delay for each check of file existance and when I call this function later from http handler - delay becomes terrible, about 14 seconds.
It seems to me that 1,33 for stat() - is inadmissible delay for such a function and 14 seconds - is out of my understanding.

### Expected Behavior
stat() on SPIFFS must work rather faster in all circumstances.
### Actual Behavior
stat() on SPIFFS tackes from 1,33 to 15 seconds acording to MCU payload or if there any other way to check file existance faster?
### Steps to repropduce

call  is_file_exist() after mounting SPIFFS and enabling WiFi and/or Ethernet, BLE, MQTT ...
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3053853/sdkconfig.zip)

### Code to reproduce this issue

bool is_file_exist(char *filename)
{
    ESP_LOGI(TAG, "Enter is_file_exist()");
    if (is_filesystem_mounted())
    {

        char *file_path = pvPortMalloc(strlen(MOUNT_PATH) + strlen(filename) + 3);

        strcpy(file_path, MOUNT_PATH);
        strcat(file_path, "/");
        strcat(file_path, filename);

        struct stat st;
        ESP_LOGI(TAG, "Before stat()");
        if (stat(file_path, &st) == 0)
        {
           ESP_LOGI(TAG, "After stat() if file exist");
            if (file_path)
            {
                vPortFree(file_path);
                file_path = NULL;
            }
            ESP_LOGI(TAG, "Exit is_file_exist()");
            return true;
        }
        else
        {
            ESP_LOGI(TAG, "After stat() if file NOT exist");
            if (file_path)
            {
                vPortFree(file_path);
                file_path = NULL;
            }
            CONFIG_LOGI("Exit is_file_exist()");
            return false;
        }
    }
   ESP_LOGI(TAG, "Exit is_file_exist()");
    return false;
}

## Debug Logs

```
ets Jun  8 2016 00:22:57

[... 146 lines cut ...]
I (322836) CONFIG: Processing value: 'ClientKeyPemFile'
I (322836) CONFIG: Enter is_file_exist()
I (322846) CONFIG: Before stat()
I (348866) CONFIG: After stat() if NOT exist
I (348866) CONFIG: Exit is_file_exist()
...............
I (348896) CONFIG: Processing value: 'CACertPemFile'
I (348906) CONFIG: Enter is_file_exist()
I (348906) CONFIG: Before stat()
I (349166) CONFIG: After stat() if exist
I (349166) CONFIG: Exit is_file_exist()
I (349166) CONFIG: Processing value: 'PvtKeyPemFile'
I (349166) CONFIG: Enter is_file_exist()
I (349166) CONFIG: Before stat()
I (376796) CONFIG: After stat() if exist
I (376796) CONFIG: Exit is_file_exist()
................
I (376886) CONFIG: Checking config finished.

 
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

igrr, 2019-04-08:
SPIFFS traverses the entire filesystem when doing `open` and `stat` operations, so they can take long time for larger partitions. See the issue in SPIFFS repository: https://github.com/pellepl/spiffs/issues/169, specifically:

> This is due to the design goal of using as little ram as possible - compared to other fs:s, there are no index tables or file trees built in memory.

Since SPIFFS does have a cache, order of files also matters. Consider 2 cases:

1) Best case, files are created in the same order as `stat`-ed. 736 read operations in 80ms:
```
I (13874) spi_flash: read   count=     736  time=   80948us  bytes=  106816
I (13875) spi_flash: write  count=       0  time=       0us  bytes=       0
I (13879) spi_flash: erase  count=       0  time=       0us  bytes=       0
```

2) Worst case, reverse order. 39k read operations in 4 seconds:
```
I (20996) spi_flash: read   count=   39272  time= 4221830us  bytes= 4741268
I (20997) spi_flash: write  count=       0  time=       0us  bytes=       0
I (21001) spi_flash: erase  count=       0  time=       0us  bytes=       0
```
<details><summary>Code used:</summary>

```c++
    test_setup();
    int file_count = 100;
    for (int i = 0; i < file_count; ++i) {
        char name[32];
        snprintf(name, sizeof(name), "%s%d.txt", filename, i);
        test_spiffs_create_file_with_text(name, "foo\n");
    }
    test_teardown();
    test_setup();
    spi_flash_reset_counters();
    for (int i = 0; i < 100; ++i) {
        char name[32];
        snprintf(name, sizeof(name), "%s%d.txt", filename, i);
        struct stat st;
        TEST_ASSERT_EQUAL(0, stat(name, &st));
        TEST_ASSERT(st.st_mode & S_IFREG);
        TEST_ASSERT_FALSE(st.st_mode & S_IFDIR);
    }
    spi_flash_dump_counters();
    test_teardown();
```
and in the second test the loop is `for (int i = 99; i >= 100; --i) {`.

`test_setup` and `test_teardown` functions are from [test_spiffs.c](https://github.com/espressif/esp-idf/blob/e7f85f1987aa9479c2dbab638ca83bcaef99be00/components/spiffs/test/test_spiffs.c#L451-L466)
</details>

In the context of IDF repository, we can't change the logic of SPIFFS. You can increase SPIFFS logical page size, which will result in fewer pages in the filesystem, and fewer read operations. With 512 byte page size, above "worst case" test results in 23k read operations. This is not orders of magnitude of improvement you are probably looking for, though.

Flash read operations are expensive on the ESP32 in dual core mode, and even more so if PSRAM support is enabled. It is possible to do some caching of flash reads, e.g. by reading whole 4kB sectors when SPIFFS only asks for 256 bytes. However this approach doesn't bring performance improvement, unfortunately. In the "worst case" test, such modification results in 13k flash read operations, but because each operation reads more data, the total time is increased to 10 seconds.

In the end, if you need better `stat` performance, it might make sense to either consider using FATFS, or to use a much smaller SPIFFS partition.

no1seman, 2019-04-08:
OK, I understand, I will try to use FATFS, but...
 
...my partition is about 2 Mbytes, only 5 files and only 9538 bytes used:
-rw-rw-r-- 1 user user 2780 mar 30 23:09 beacon.json
-rw-r--r-- 1 user user 1150 oct  7  2018 favicon.ico
-rw-r--r-- 1 user user 1159 nov 22 08:26 http_cert.pem
-rw-r--r-- 1 user user 1704 nov 22 08:26 http_key.pem
-rw-rw-r-- 1 user user  959 nov 27 11:03 index.html

If I'm right stat() needs to read only 5 blocks (maimum) to find a file and all information to fill stat structure. I'm really confused that reading of 5 blocks may long 15 seconds. If it is so there no ways to use SPIFFS on ESP32 at all for any purposes?

Thanks

closed 2019-04-08 by no1seman

igrr, 2019-04-08:
> If I'm right stat() needs to read only 5 blocks (maximum) to find a file and all information to fill stat structure. I'm really confused that reading of 5 blocks may long 15 seconds.

@no1seman Please try enabling `CONFIG_SPI_FLASH_ENABLE_COUNTERS` in menuconfig, call `spi_flash_reset_counters()` before calling `stat`, and call `spi_flash_dump_counters()` after calling `stat`. This will print the number of read operations performed by SPIFFS.