espressif/esp-idf #3278: Get SD card free space not updating

gunarkroeger, 2019-04-08:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version v3.2-rc-9-g70811110f
- Build System: Make
- Operating System:    Windows
- Power Supply:         external 5V

## Problem Description

I was trying to monitor if the SD card connected through SPI had been removed or if it was getting close to being full through this function:

[Codebox=c file=Untitled.c]
bool SD_getFreeSpace(uint32_t *tot, uint32_t *free)
{
	FATFS *fs;
	DWORD fre_clust, fre_sect, tot_sect;

	/* Get volume information and free clusters of drive 0 */
	if(f_getfree("0:", &fre_clust, &fs) == FR_OK)
	{
		/* Get total sectors and free sectors */
		tot_sect = (fs->n_fatent - 2) * fs->csize;
		fre_sect = fre_clust * fs->csize;

		*tot = tot_sect / 2;
		*free = fre_sect / 2;

		/* Print the free space (assuming 512 bytes/sector) */
		ESP_LOGD(TAG, "%10lu KiB total drive space. %10lu KiB available.", *tot, *free);

		return true;
	}
	return false;
}
[/Codebox]

But to my suprise it does not work as expected

### Expected Behavior
Calling the function returns current total and free space in the SD card.

### Actual Behavior
Function returns values from the time the card was mounted, and doesn't get updated, even if the card is removed
### Steps to repropduce

1. Use SD example from idf as a base line to mount and write files.
2. Call this function, check values
3. Write big files
4. Call this function, check values
5. Remove SD card
6. Call this function, check values

Values will always be the same
https://www.esp32.com/viewtopic.php?f=2&t=9914&sid=78355a34d5d8da75d1400404cdb85e48

igrr, 2019-04-08:
Hi @gunarkroeger, are you using the card detect pin of the SD card slot to check if the card is present? Or you wish to somehow check for card presence without using a separate card detect pin?

gunarkroeger, 2019-04-12:
Is there a workaround for this bug to check SD free space without unmounting and mounting again while it is not resolved? I've worked around checking if card is present by periodically opening and closing a file.

igrr, 2019-04-16:
@gunarkroeger I've tested your code snippet, but I was not able to reproduce the issue that the available space reported by this function doesn't change after creating some files. My test code looks like this:

```c++
void test_fatfs_disk_size(const char* filename)
{
    uint32_t tot_0, tot_1, free_0, free_1;

    unlink(filename);
    TEST_ASSERT(SD_getFreeSpace(&tot_0, &free_0));

    /* create a 64kB file */
    const size_t file_size = 65536;
    FILE* f = fopen(filename, "wb");
    TEST_ASSERT_NOT_NULL(f);
    uint32_t val = UINT32_MAX;
    for (size_t i = 0; i < file_size / sizeof(val); ++i) {
        ssize_t n = fwrite(&val, sizeof(val), 1, f);
        TEST_ASSERT_EQUAL(1, n);
    }
    fclose(f);

    TEST_ASSERT(SD_getFreeSpace(&tot_1, &free_1));
}
```

Testing this with FATFS on SD card, I get the following (logged from `SD_getFreeSpace` function):
```
  15554288 KiB total drive space.   15551488 KiB available.
  15554288 KiB total drive space.   15551424 KiB available.
```

So the difference in "available" size matches the size of the file written.

gunarkroeger, 2019-04-16:
@igrr You are correct. I made a mistake while testing.
Values are updating for size changes.
But physically removing the SD card and calling SD_getFreeSpace does not return an error or change the sizes to 0. Could you gave a look into this as well?
thanks

igrr, 2019-04-16:
FATFS library does not actually access the card when f_getfree function is called (in most cases). It simply returns values based on the FAT state in memory. It seems that FATFS APIs do not provide a specific function to "check if media is present". As you have mentioned, it is possible to abuse one of the functions which perform disk write operations, to do that.

closed 2019-04-16 by gunarkroeger