espressif/esp-idf #3279: Unable to call dbg_lwip_stats_show() function

zhivko, 2019-04-08:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (v2]
- Module or chip used:  ESP32-WROVER
- IDF version v3.2-beta3-60-gf49e2afe9
- Build System:         Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux subsystem on Windows 10
- Power Supply:         USB

## Problem Description

How I properly include headers so I could call function:
dbg_lwip_stats_show()
from my Server.cpp?

I tried with including:
```
#include "debug/lwip_debug.h"
```
and adding below line inside debug/lwip_debug.h
```
void dbg_lwip_stats_show(void);
```

But after
```
make all -j7
```

I still get below error while compiling my esp-idf project.

```
Toolchain path: /home/klemen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS  FS HTTPClient  NetBIOS Preferences SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
Python requirements from /home/klemen/esp/esp-idf/requirements.txt are satisfied.
LD build/DoubleLifter.elf
/home/klemen/esp/DoubleLifter/build/main/libmain.a(Server.o):(.literal._Z6myLoopv+0x4c): undefined reference to `dbg_lwip_stats_show()'
/home/klemen/esp/DoubleLifter/build/main/libmain.a(Server.o): In function `myLoop()':
/home/klemen/esp/DoubleLifter/main/Server.cpp:2124: undefined reference to `dbg_lwip_stats_show()'
collect2: error: ld returned 1 exit status
/home/klemen/esp/esp-idf/make/project.mk:460: recipe for target '/home/klemen/esp/DoubleLifter/build/DoubleLifter.elf' failed
make: *** [/home/klemen/esp/DoubleLifter/build/DoubleLifter.elf] Error 1

### Expected Behavior

Correctly compiled firmware.

### Actual Behavior
klemen@DESKTOP-I75PLBB:~/esp/DoubleLifter$ make all flash -j7
Toolchain path: /home/klemen/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Arduino libraries in build: ArduinoOTA AsyncUDP AzureIoT BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS  FS HTTPClient  NetBIOS Preferences SD SD_MMC SPI SPIFFS SimpleBLE Ticker Update WebServer WiFi WiFiClientSecure Wire
CXX build/main/Server.o
Python requirements from /home/klemen/esp/esp-idf/requirements.txt are satisfied.
AR build/main/libmain.a
LD build/DoubleLifter.elf
/home/klemen/esp/DoubleLifter/build/main/libmain.a(Server.o):(.literal._Z6myLoopv+0x4c): undefined reference to `dbg_lwip_stats_show()'
/home/klemen/esp/DoubleLifter/build/main/libmain.a(Server.o): In function `myLoop()':
/home/klemen/esp/DoubleLifter/main/Server.cpp:2124: undefined reference to `dbg_lwip_stats_show()'
collect2: error: ld returned 1 exit status
/home/klemen/esp/esp-idf/make/project.mk:460: recipe for target '/home/klemen/esp/DoubleLifter/build/DoubleLifter.elf' failed
make: *** [/home/klemen/esp/DoubleLifter/build/DoubleLifter.elf] Error 1

### Steps to reproduce
### Code to reproduce this issue
https://github.com/zhivko/DoubleLifter/blob/master/main/Server.cpp#L2020

liuzfesp, 2019-07-16:
You can enable it via menuconfig: make menuconfig => component config => LWIP => Enable LWIP statistics.

Alvin1Zhang, 2019-07-16:
@zhivko Thanks for reporting the issue, feel free to reopen the issue if it still happens. Thanks.

closed 2019-07-16 by Alvin1Zhang