espressif/esp-idf #3280: Wrong path? Syntax error?

jaumemirallesisern, 2019-04-08:
Trying to run "hello world" example.
ubuntu version: 14.04

Followed standard setup for linux here:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html

Everything goes ok until I try to execute '**make flash**' inside 'hello world' folder. It returns:

> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
> WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
> WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
> CC build/bootloader/bootloader_support/src/bootloader_clock.o
> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
> make[2]: *** [src/bootloader_clock.o] Error 2
> make[1]: *** [component-bootloader_support-build] Error 2
> make: *** [/home/jaume/esp/hello_world/build/bootloader/bootloader.bin] Error 2

**printenv PATH returns:**

> /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/jvm/java-8-oracle/bin:/usr/lib/jvm/java-8-oracle/db/bin:/usr/lib/jvm/java-8-oracle/jre/bin:/home/jaume/esp/xtensa-esp32-elf/bin

The directory **/home/jaume/esp/xtensa-esp32-elf/bin** contains among other files:

> xtensa-esp32-elf-gcc

**printenv IDF_PATH returns**:

> /home/jaume/esp/esp-idf

The directory **/home/jaume/esp/esp-idf** looks like:

> add_path.sh  components  CONTRIBUTING.rst  docs  examples  Kconfig  LICENSE  make  README.md  requirements.txt  tools

I don't really understand the errors as the PATH 'seems' to be ok.

thanks

projectgus, 2019-04-09:
Hi @jaumemirallesisern ,

Thanks for reporting this. Something unusual seems to be happening!

Can you please try running the following two commands and let us know what they say:

```
xtensa-esp32-elf-gcc -v
```

```
file /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```

(You may need to run `sudo apt-get install file` if your Ubuntu system doesn't have the "file" utility installed.)

Thanks.

jaumemirallesisern, 2019-04-09:
Thanks for your answer. Let's see:

executing **xtensa-esp32-elf-gcc -v** returns:

> bash: /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: cannot execute binary file: Exec format error

Executing **file /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc** returns:

> /home/jaume/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF 64-bit LSB  executable, x86-64, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.32, BuildID[sha1]=81a0b55c5c0c1bae1bb9d9ef2ffafc0a054bedaf, stripped

projectgus, 2019-04-09:
Hi @jaumemirallesisern,

Thanks. Is it possible you have 32-bit rather than 64-bit Ubuntu? (running `uname -a` is one way to tell this).

If you have 32-bit Linux (x86 not x86_64) then you'll need the 32-bit Linux toolchain not the 64-bit one.

jaumemirallesisern, 2019-04-09:
Ops! Yes it is.
Silly me, my mind was thinking of a remote server when I decided the file, sorry.
Thank you very much for your patience

projectgus, 2019-05-16:
Hi @jaumemirallesisern ,

Glad we got to the bottom of it. I'm going to close this as I assume it's working with the right toolchain, now.

closed 2019-05-16 by projectgus