espressif/esp-idf #3281: BLE: App falls down with BT_BTC: btc_gattc_cback transfer failed

ilyamordasov, 2019-04-09:
Time to time my app falls down with this error, it happens once in 3-4 times. Looking at heap, ~ 114144 bytes are free, and when BT starts ~ 5kb are free (I tried, BT works good if more then 1 kb are available)

It happens while I try to open BT device using its address.

Some logs:
```
I (23622) APP: GATTC_PROFILE_EVENT_HANDLER: unknown event: 0
W (23632) APP: Attempting to call esp_ble_gatt_set_local_mtu
Bluetooth start 1
I (23632) APP: connect to the remote device.
E (23862) BT_BTC: btc_gattc_cback transfer failed

I (23872) APP: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (23872) APP: REMOTE BDA:
I (23872) APP: e5 fb 01 09 f7 b4 
E (24722) BT_BTC: btc_gattc_cback transfer failed
W (119432) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
I (119432) APP: ESP_GATTC_DISCONNECT_EVT, reason = 19
I (119442) APP: ESP_GATT_CONN_NONE: 0
I (119442) APP: GATTC_PROFILE_EVENT_HANDLER: unknown event: 5
I (137682) event: station ip lost
I (137682) network: event 8
```

or  when I'm attempting to write bytes into characteristic, sometimes it suspends on this one
```
I (24312) APP: ATTEMTING TO WRITE TO CHARACTERISTIC in HANDLE 0x000e
I (24312) APP: 55 00 ff df 24 0e c6 94 d1 97 43 aa 
I (24342) APP: write char ok
I (24342) APP: ATTEMTING TO WRITE TO CHARACTERISTIC in HANDLE 0x000c
I (24342) APP: 01 00 
I (24372) APP: write char ok
I (24372) APP: ATTEMTING TO WRITE TO CHARACTERISTIC in HANDLE 0x000e
I (24372) APP: 55 01 01 aa 
I (140082) event: station ip lost
I (140082) network: event 8
```

Weijian-Espressif, 2019-04-09:
@ilyamordasov, please modify `BTC_TRACE_ERROR("%s transfer failed\n", __func__);` to `BTC_TRACE_ERROR("%s transfer failed %d\n", __func__, ret);`in` btc_gattc_cback()`, and show me the log.

ilyamordasov, 2019-04-09:
@Weijian-Espressif 
```
I (72352) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (72362) APP: Attempting to call esp_bt_controller_enable
W (72532) APP: Attempting to call esp_bluedroid_init
W (72542) APP: Attempting to call esp_bluedroid_enable
W (74602) APP: Attempting to call esp_ble_gap_register_callback
W (74602) APP: Attempting to call esp_ble_gattc_register_callback
W (74602) APP: Attempting to call esp_ble_gattc_app_register
I (74612) APP: GATTC_PROFILE_EVENT_HANDLER: unknown event: 0
W (74612) APP: Attempting to call esp_ble_gatt_set_local_mtu
Bluetooth start 1
I (74622) APP: connect to the remote device.
E (74702) BT_BTC: btc_gattc_cback transfer failed 3

I (74712) APP: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (74712) APP: REMOTE BDA:
I (74712) APP: e5 fb 01 09 f7 b4 
E (75362) BT_BTC: btc_gattc_cback transfer failed 3
```

And in the other case

```
W (32932) APP: Attempting to call esp_ble_gatt_set_local_mtu
Bluetooth start 1
I (32942) APP: connect to the remote device.
W (33292) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e
I (33292) APP: ESP_GATTC_DISCONNECT_EVT, reason = 62
I (33292) APP: ESP_GATT_CONN_NONE: 0
E (33302) APP: open failed, status 133
I (135562) event: station ip lost
I (135562) network: event 8
```

Those two cases have only different bytes to be sent (the 1st portion turns on my device the 2nd one turns off)

Weijian-Espressif, 2019-04-10:
@ilyamordasov, ret = 3. it is BT_STATUS_NOMEM, please refer to `btc_transfer_context()` in btc_task.c. 
```
lmsg.arg = (void *)osi_malloc(arg_len);
        if (lmsg.arg == NULL) {
            return BT_STATUS_NOMEM;
        }
```
This is because the memory is not enough.

ilyamordasov, 2019-04-10:
Ok, thank you. I had thoughts it could be memory leak, but I was checking free heap after and before BT call. Before BT enable I have ~110 kB, before BT connect ~5 kB. My tests showed while free memory is greater then 1 kB I can use BT. When my BT sent all data I disconnect it close Bluedroid etc, deinit them and get the same 110 kB

closed 2019-04-12 by ilyamordasov