espressif/esp-idf #3283: Problem: ESP32 panic’ed when running ov2640(camera) initialization

xloc, 2019-04-09:
## Environment

- Development Kit: none
- Module or chip used: ESP32-PICO-D4
- IDF version: v3.0.7
- Build System: Make
- Compiler version: 
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: macOS
- Power Supply: USB.  
Switch mode power supply included; It caused a problem when running wifi scan example, but the problem has been solved by adding a large (220uF) capacitor near the chip

## Keywords:
ESP32-PICO-D4; OV2640 camera; panic’ed (IlligalInstruction); SCCB; self-made 4-layer PCB

## Problem Description

I designed a PCB for a project including video streaming functionality, and the chip I used was ESP32-PICO-D4.
Since it will be even harder to build streaming from scratch, I referenced the design of M5CAM. Specifically, I kept an almost identical connection with the camera of m5cam in my schematic for avoiding connection error.
> Their project can be found there: 
> [https://github.com/1zlab/1ZLAB_ESP32_Wifi_Camera](https://github.com/1zlab/1ZLAB_ESP32_Wifi_Camera)
> [https://github.com/m5stack/esp32-cam-demo](https://github.com/m5stack/esp32-cam-demo)

Only the following is different:
1. I used ESP32-PICO-D4 rather than the chip used in m5cam which requires external flash ROM.
2. I connected two extra devices to the SCCB interface (the control interface of ov2640; seemingly identical with i2c)
As a result, I shall use the code of m5cam to test if my hardware design is correct.

However, it turns out to be problematic. T-T
I uploaded the program, the program which worked on the official board of m5stack, to my board. And it simply does not work, and outputs register dump messages. Even worse, the problem is quite weird.

>PS: there is a little bug in m5cam software that it polls the devices attach to the bus, from 1 to 127, and take the first acknowledged device as the camera. It works fine when there is only one camera(device) on the bus, but I have two extra, with one of them address=0x20, ahead of ov2640 address=0x30, so being selected erroneously. This small problem has been solved, and I will only discuss the 'correct' version of the code in the following.

### Expected Behavior
The code works on my board

### Actual Behavior
Here are my debug steps
##### Step 1: Use 'unmodified' firmware. The dump message is like:
[image]

There isn’t much valuable information in this dump
As shown in the stack backtrace, the problem is at the `app_main` function. (without any surprise)
The line number point to `if (camera_model == CAMERA_OV7725) {`which should not be an `IllegalInstrution`, therefore I recognize that it reports a wrong line number. (the ESP official mentioned it in the debug document that the line number might be incorrect, but the function name is always correct)

##### Step 2: I used many log messages to locate where the problem is from.
I found the problem is in the reset function, and should inside the while loop.
```c
// at project_root/components/camera/ov2640.c
static const char *TAG = "ov2640";
static int reset(sensor_t *sensor)
{
    int i=0;
    const uint8_t (*regs)[2];

    /* Reset all registers */
    SCCB_Write(sensor->slv_addr, BANK_SEL, BANK_SEL_SENSOR);
    SCCB_Write(sensor->slv_addr, COM7, COM7_SRST);

    /* delay n ms */
    delay(10);

    i = 0;
    regs = default_regs;
    /* Write initial regsiters */
    ESP_LOGW(TAG, "Start writing registers");
    while (regs[i][0]) {
        SCCB_Write(sensor->slv_addr, regs[i][0], regs[i][1]);
        i++;
    }
    ESP_LOGW(TAG, "End of writing registers");

    i = 0;
    regs = svga_regs;
    /* Write DSP input regsiters */
    while (regs[i][0]) {
        SCCB_Write(sensor->slv_addr, regs[i][0], regs[i][1]);
        i++;
    }

    return 0;
}
```
[image]

The panic type is still `IllegalInstruction`, but the problem is strange. The error is from the last log call (line 410. hopefully the line number is correct this time). And I have to think that there are some operations in the while loop changed something and cause the log failure.

##### Step 3: More logs are added into the while loop to find exactly where the error comes from.
```c
    i = 0;
    regs = default_regs;
    /* Write initial regsiters */
    ESP_LOGW(TAG, "Start writing registers");
    while (regs[i][0]) {
        ESP_LOGW(TAG, "i=%d", i);
        SCCB_Write(sensor->slv_addr, regs[i][0], regs[i][1]);
        ESP_LOGW(TAG, "writing done");
        i++;
        ESP_LOGW(TAG, "increment done");
    }
    ESP_LOGW(TAG, "End of writing registers");
```
[image]

##### Step 4: Remove the last two logging statement
```c
    i = 0;
    regs = default_regs;
    /* Write initial regsiters */
    ESP_LOGW(TAG, "Start writing registers");
    while (regs[i][0]) {
        ESP_LOGW(TAG, "i=%d", i);
        SCCB_Write(sensor->slv_addr, regs[i][0], regs[i][1]);
        i++;
    }
    ESP_LOGW(TAG, "End of writing registers");
```
[image]

Start reboot and reset loop immediately. I will mention it later. It occurs after each dump, and it is another layer of weirdness.

##### Step 5: I wonder the error is caused by high writing speed. So, I add delay to slow it down.
```c
    i = 0;
    regs = default_regs;
    /* Write initial regsiters */
    ESP_LOGW(TAG, "Start writing registers");
    while (regs[i][0]) {
        ESP_LOGW(TAG, "i=%d", i);
        SCCB_Write(sensor->slv_addr, regs[i][0], regs[i][1]);
        i++;
        delay(100);
    }
    ESP_LOGW(TAG, "End of writing registers");
```
delay is defined as
```c
void delay(int millis) {
	vTaskDelay(millis / portTICK_PERIOD_MS);
}
```
there are two random dump cases
[image]
[image]

##### Description of the after-dump situation:
THIS IS WEIRD. IT HAPPENS AFTER EACH OF THE CASES STATED BEFORE.
After the dump, the program will continuously output the following message and will not run correctly even after reset.

[image]
[image]

The doc says getting this message is due to flash encryption, which I haven’t configured any.

By the way, I am able to reprogram the chip (or I will go crazy) by power off the esp32 and power it on just after the uploader start to connect to the chip. And It is also the only way I could rerun the certain program.

##### I guess the problem might come from the following places:
1. The esp32-pico chip may have broken due to my hand soldering (manual reflow with stencil). I start to less believe it since the GPIO manipulation and WIFI-scan program seem working fine.
2. Although the SCCB interface can be treated as i2c, m5stack uses a self-defined sccb driver, which works on their own board. Might it be a problem?
3. It may be related to privileged memory access, my guess. I saw `ProhibitedLoad/Store` panic during the camera reset function(the function mentioned) debug, but I forgot the exact condition.
4. Is there any initialization operation that is required for a raw chip? I missed it completely if there is any.

### Steps to repropduce

The problem related too much to my PCB, so it might be hard to reproduce the error.

### Code to reproduce this issue

inlined with the problem description.

## Debug Logs

inlined with the problem description.

---
I stuck at this problem for 5 days. Please, if you know any reason or have any clue that what happens to my hardware, feel absolutely free to add a comment. Thanks a lot.

igrr, 2019-04-09:
This usually happens if some peripheral is connected to SPI flash pins, or there is a short between one of the SPI flash pins an a different signal. PICO uses GPIOs 6, 7, 8, 11, 16, 17 as SPI flash signals internally, so make sure you don't have anything connected to them. If in doubt, please post the schematic of your board. Also please inspect your soldering for shorts between the SPI flash pins listed above and surrounding pins in the package.

By the way, since your issue is hardware related, it might be more suited for the esp32.com forum than for the ESP-IDF issue tracker.

xloc, 2019-04-09:
Sorry about posting it to the wrong place, but I had some trouble when uploading the images to esp32.com

I did not connect anything to GPIO6/7/8/11, and GPIO16 is connected to the PWM input of Motor which is high impedance from the perspective of esp32, GPIO17 is connected to a digital input which kept the same with the original M5CAM project. So, it should be okay to use this connection.

Here is my schematic
[image]

I will check the hardware connection to my best effort, though it might be hard(DFN package + no test point of GPIO6/7/8/11)

Besides, could you please talk about more details of the symptom when misusing the SPI pins? 

And thank you very much for your reply.

igrr, 2019-04-09:
The design you are referencing used ESP32-D2WD, which uses GPIOs 6-11 to connect the SPI Flash. So using GPIO 16 and 17 there was fine. However GPIO 16 and 17 are connected to the SPI flash inside the ESP32-PICO-D4 package. This is mentioned in the pin list of the PICO-D4 datasheet. You can't use these pins for other purposes (input or output).

xloc, 2019-04-10:
Thank you, I will disconnect it in my next PCB.

However, I haven't manipulated these two pins in my program. The program didn't even reach the line setting I/O directions or setting the level of these two pins. So, the program should not go wrong, correct? If you have any other clue about what might cause this problem, please inform me.

Anyway, I will cut my PCB to physically disconnect these two pins to esp32-pico, and see if the problem solves. I will report my test result later. Thanks

igrr, 2019-04-10:
GPIO17 is connected to one of the camera outputs. It is also connected to an SPI flash pin. When camera is enabled and starts driving the pin, the communication between SPI flash and ESP32 on the same pin is disturbed, hence ESP32 can not read instructions from flash successfully, and crashes.

xloc, 2019-04-11:
Okay, understood. I will report later the bug fixing result. Thanks

xloc, 2019-04-13:
Great. The problem solved, thank you very much for your support.

The CAM_Y2 which connects to GPIO17 might change its level during the SCCB setting and ruin the SPI Flash ROM signal. I cut the wire on PCB then the program works.

Though it is now frequently brownout due to my switching mode power supply, it is another story. I will post it on esp32.com

Thank you

closed 2019-04-15 by Alvin1Zhang