espressif/esp-idf #3284: Make environment exits with error when secure boot key is absent.

michelesolytic, 2019-04-09:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v3.1.3
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:  Linux (Fedora 29)
- Power Supply: Other

## Problem Description
When attempting to enable Secure Boot according to the API guide at https://docs.espressif.com/projects/esp-idf/en/stable/security/secure-boot.html and when the signing key is not found, the build environment should prompt a message with the command necessary to generate such key. However, an error occurs instead and the make environment quits with an error. See attached log.

## Debug Logs

```
 [michele@michele-macbook firmware]$ make
 GENCONFIG
 #
 # configuration written to /home/michele/git/funshine/firmware/sdkconfig
 #
 Building partitions from /home/michele/git/funshine/firmware/partitions.csv...
 Need to generate secure boot signing key.
 One way is to run this command:
 /bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
 /bin/sh: -c: line 1: syntax error: unexpected end of file
 make: *** [/home/michele/git/esp-idf/components/bootloader_support/Makefile.projbuild:4: run/media/michele/Solytic\] Error 1
```
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3058310/sdkconfig.txt)

michelesolytic, 2019-04-09:
Didn't look in the python script, but after some tries it looks like escaped spaces in file paths are not interpreted correctly. For example /home/michele/my\ device/key.pem does not work, but /home/michele/my_device/key.pem does.

projectgus, 2019-06-05:
Hi @michelesolytic ,

Sorry noone got back to you about this earlier. Thanks for being patient.

Right now spaces in paths (for projects, IDF_PATH, or other file paths) aren't supported by the build system. We have this on our list as a future feature to implement, but there's no ETA at this stage.

If removing the space fixed the problem, can we close this issue?

michelesolytic, 2019-06-05:
Thanks for the answer!
Yes indeed I worked around it by just using underscores _ rather than spaces. I look forward to improvements :)

closed 2019-06-05 by michelesolytic