espressif/esp-idf #3286: How to move bin file from arduino-ide to msys32 ? OR how to build app with arduino-esp32 as component?

BinaryWorlds, 2019-04-09:
I wrote a bigger project in arduino-ide and esp32. I need to enable secure boot and use flash encryption. Arduino-ide doesn't support this.

I see 2 ways:

1) export binary from arduino-ide to toolchain and encrypt this. It's possible but i dont know how[use bin from first]. I found some info here: https://github.com/espressif/arduino-esp32/issues/1387

2) build app in toolchain. But it request this repository: https://github.com/espressif/arduino-esp32 as a component to esp-idf. https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

I use different version of this and still take error like here: (when compile easy project)

C:/msys32/home/user/esp/test/components/arduino/cores/esp32/esp32-hal-time.c:16:28: fatal error: lwip/apps/sntp.h: No such file or directory compilation terminated. make1: * [/home/user/esp/esp-idf/make/component_wrapper.mk:286: cores/esp32/esp32-hal-time.o] Błąd 1 make: * [C:/msys32/home/user/esp/esp-idf/make/project.mk:468: component-arduino-build] Błąd 2

In addition, this is my first contact with this application. So im not sure how select only needed components. Currently, many unnecessary libraries are being built.

Thank you for any help and forgive me for the lack of knowledge. Not everything is clear to me.

My main problem is the easiest way to secure the app on the esp board.

DarvosBR, 2019-04-12:
I can help you with this!
If u wanna use Arduino.h (the framework) on esp-idf, just follow those instructions!
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

if you wanna use esp-idf code on arduino, 
`bool loopTaskWDTEnabled;

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
        loop();
    }
}

extern "C" void app_main()
{
    loopTaskWDTEnabled = false;
    initArduino();
    xTaskCreatePinnedToCore(loopTask, "loopTask", 4096, NULL, 1, &loopTaskHandle, 1);
    
    
    
}`
Do something like this.

projectgus, 2020-12-01:
Closing as this is not an ESP-IDF issue, and Nathan has kindly provided an outline of how to do it.

closed 2020-12-01 by projectgus