espressif/esp-idf #3287: A stack overflow in task btuT has been detected

PegasYura, 2019-04-10:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (ESP-IDF Pre-release v3.2-rc):  //boot: ESP-IDF v3.2-rc-dirty 2nd stage bootloader
- Build System:         [Make]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

Stack checking was constantly turned on in my project.
I started working with BLUFI
So, when I run the BLUFI example with the standard settings sdkconfig, everything is fine.
As soon as I turn on:
CONFIG_STACK_CHECK_NONE =
CONFIG_STACK_CHECK_NORM =
CONFIG_STACK_CHECK_STRONG =
CONFIG_STACK_CHECK_ALL = y
CONFIG_STACK_CHECK = y
Every time I try to enable encryption in BLUFI, a crash occurs.

## Debug Logs

```
I (895) BLUFI_EXAMPLE: BLUFI VERSION 0102

I (905) BLUFI_EXAMPLE: BLUFI init finish

I (18335) BLUFI_EXAMPLE: BLUFI ble connect

***ERROR*** A stack overflow in task btuT has been detected.
abort() was called at PC 0x40096bc1 on core 0
0x40096bc1: vApplicationStackOverflowHook at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/esp32/panic.c:708

Backtrace: 0x400968e3:0x3ffd7ca0 0x40096b9c:0x3ffd7cd0 0x40096bc1:0x3ffd7d00 0x4009369c:0x3ffd7d30 0x40095610:0x3ffd7d60 0x400955c6:0x00000001
0x400968e3: invoke_abort at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/esp32/panic.c:708

0x40096b9c: abort at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/esp32/panic.c:708

0x40096bc1: vApplicationStackOverflowHook at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/esp32/panic.c:708

0x4009369c: vTaskSwitchContext at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/freertos/tasks.c:3538

0x40095610: _frxt_dispatch at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/freertos/portasm.S:406

0x400955c6: _frxt_int_exit at /home/osboxes/esp/hcm360_fw/HCM360_WiFi_IR_HW_V5.1/esp-idf/components/freertos/portasm.S:206

Rebooting...
ets Jun  8 2016 00:22:57
```

chegewara, 2019-04-10:
https://github.com/espressif/esp-idf/issues/2503

PegasYura, 2019-04-10:
Thank you, I increased the task stack
#define BTU_TASK_STACK_SIZE             (6144 + BT_TASK_EXTRA_STACK_SIZE)

closed 2019-04-10 by PegasYura