espressif/esp-idf #3288: [question] Is there a way to add a partition during runtime?

farizadp, 2019-04-10:
We need to add coredump partition type into our current partition table (for debugging a reboot issue possibly caused by a segfault). The thing is that the device is already in production, and the only access to it is just internet (i.e. no UART).

Is there such a way to add coredump partition during runtime? or perhaps any alternative to get the coredump after reboot not via UART?

Regards

negativekelvin, 2019-04-10:
It's not recommended to change the bootloader or partition table from a running app.  If you have unpartitioned space available it would be better to modify espcoredump to use that or maybe to use an inactive ota partition.

farizadp, 2019-04-11:
Thanks, modifying the espcoredump indeed sounds better. I'll give a try

Alvin1Zhang, 2019-05-16:
@farizadp Thanks for reporting the issue. Could you help share if any updates for the issue? Has your issue been fixed? Thanks.

farizadp, 2019-06-05:
@Alvin1Zhang : we haven't had time to try changing the espcoredump. I'll update this thread once we've tried changing the espcoredump

lrodenasSistel, 2019-08-30:
For future users who find this: dynamically changing the partition table is doable.
I explain how we did it in this other topic: https://www.esp32.com/viewtopic.php?f=2&t=11482&p=48589#p48589

Alvin1Zhang, 2020-04-29:
@farizadp Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-04-29 by Alvin1Zhang