espressif/esp-idf #3289: Modbus controller incorrectly set a read coil event as MB_EVENT_COILS_WR

domoticalabs-dev, 2019-04-10:
Into linked code line, and the next one (426), the event is set in a wrong way since in this case the variable eMode is equal to MB_REG_READ.

https://github.com/espressif/esp-idf/blob/c064e008424d6f87cc78ca93ed572d4b2ba346d9/components/freemodbus/modbus_controller/mbcontroller.c#L425

The proposed solution is to change, in both lines, the value MB_EVENT_COILS_WR in MB_EVENT_COILS_RD.

alisitsyn, 2019-04-10:
Hi domoticalabs-dev,

Thank you for reporting this. The change will be applied into current release and merged as soon as possible.

--
Alex

domoticalabs-dev, 2019-04-11:
Is there any plan to implement a TCP slave too on the future? It seems there is not in the library, so we've tried to implement a simple example of it, still we don't have enough resources to make it completely working and we don't have any plan of doing it in the future. 
Could you be interested in the work (very minimal) we've done so far? We have no problem in sharing it, still it's far from being ready. If you needs something to start with, we can share it with no problems.

alisitsyn, 2019-04-11:
Hi domoticalabs-dev,
Yes, there are plans to implement Modbus TCP slave and master. The work in progress common interface for master and slave is already implemented and can be found here: https://www.esp32.com/viewtopic.php?f=13&t=9527&hilit=master+modbus
This code will be merged soon. I already implemented the work in progress version of Modbus TCP slave and TCP master based on this interface code but complete implementation is postponed.
However it would be interesting for me to see your code too. Thank you.

alisitsyn, 2019-04-11:
domoticalabs-dev, 
I am not sure if I can request your code and use it due to some regulations. Also I have already implemented code. Thank you for your proposal.

domoticalabs-dev, 2019-04-12:
As you prefer, if you change your mind just ask me. I can also open a pull request if you need it and if it can help you.
I'll leave the issue open so you can close it when you merge it in the next release.

closed 2019-04-17 by igrr