espressif/esp-idf #3291: (vPortAssertIfInISR)- assert failed

chegewara, 2019-04-11:
Hi, im not sure i should open this issue here or in AWSFreeRTOS repo, but logs are suggesting its esp-idf problem (or issue in my code). Its bigger app, with wifi + ble + loboris TFT driver on m5stack. I didnt have this issue for some time until now. 

```
I (737) WIFI: SYSTEM_EVENT_STA_START
1 59 [main] [INFO ][vStartBLEScanDemo][59] start scan demo
2 59 [Btc_task] [INFO ][prvRegisterCb][59] register client callback
3 59 [Btc_task] [INFO ][prvScanParamSetCompltCb][59] stop scan,gattIf: 0, status: 0
4 59 [Btc_task] [INFO ][prvStartScanCb][59] start scan, status: 0
5 59 [ScanDemo] [INFO ][prvScanDemoTask][59] start ble scan

==============================
TFT display DEMO, LoBo 11/2017
==============================
Pins used: miso=19, mosi=23, sck=18, cs=14
==============================

SPI: display device added to spi bus (2)
SPI: attached display device, speed=8000000
SPI: bus uses native pins: true
SPI: display init...
I (977) wifi: n:2 0, o:1 0, ap:255 255, sta:2 0, prof:1
I (1647) wifi: state: init -> auth (b0)
I (1647) wifi: state: auth -> assoc (0)
I (1647) wifi: state: assoc -> run (10)
core=0 port_interruptNesting=1

E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:315 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x40092a3b on core 0
0x40092a3b: vPortAssertIfInISR at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Backtrace: 0x40096448:0x3ffc0560 0x40096617:0x3ffc0580 0x40092a3b:0x3ffc05a0 0x40093496:0x3ffc05c0 0x4012f64a:0x3ffc05e0 0x40082162:0x3ffc0600 0x400938de:0x00000000
0x40096448: invoke_abort at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x40096617: abort at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x40092a3b: vPortAssertIfInISR at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

0x40093496: vTaskEnterCritical at E:/amazon-freertos/lib/FreeRTOS/tasks.c:4944

0x4012f64a: rtc_isr at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/driver/rtc_module.c:1910

0x40082162: _xt_lowint1 at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x400938de: prvIdleTask at E:/amazon-freertos/lib/FreeRTOS/tasks.c:4944
```
```
I (3677) wifi: connected with ssid, channel 2
I (3677) wifi: pm start, type: 1

I (3677) WIFI: SYSTEM_EVENT_STA_CONNECTED
core=0 port_interruptNesting=1

E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:315 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x40092a3b on core 0
0x40092a3b: vPortAssertIfInISR at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

Backtrace: 0x40096448:0x3ffc0560 0x40096617:0x3ffc0580 0x40092a3b:0x3ffc05a0 0x40093496:0x3ffc05c0 0x4012f64a:0x3ffc05e0 0x40082162:0x3ffc0600 0x40133671:0x00000000
0x40096448: invoke_abort at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x40096617: abort at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/panic.c:649

0x40092a3b: vPortAssertIfInISR at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

0x40093496: vTaskEnterCritical at E:/amazon-freertos/lib/FreeRTOS/tasks.c:4944

0x4012f64a: rtc_isr at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/driver/rtc_module.c:1910

0x40082162: _xt_lowint1 at E:/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x40133671: esp_vApplicationIdleHook at E:/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/freertos_hooks.c:51 (discriminator 1)
```
This crash does not happen very often and it can be 1 crash and after reboot it starts or it can crash 15-20 times in a row before it runs.

chegewara, 2019-04-11:
I found part of code that i added recently and is related to this crash. It is freertos software timer, when i am not using software timer then all is ok, but with software timer im having above crash.

mahavirj, 2019-04-11:
@chegewara Can you please confirm if your `rtc_isr` looks like one from https://github.com/aws/amazon-freertos/blob/master/lib/third_party/mcu_vendor/espressif/esp-idf/components/driver/rtc_module.c#L1929 (This was recently fixed to use correct critical section APIs from ISR context)? If yes, any custom handler registered with rtc isr in your application (using `rtc_isr_register`)?

Interim, we have internal change request in development queue that will ensure that all such instances of incorrect critical section APIs from ISR context are fixed, it will be merged soon.

chegewara, 2019-04-11:
Hi @mahavirj 
code i pulled from ble-beta branch looks the same:

current master code:
```
static void rtc_isr(void* arg)
{
    uint32_t status = REG_READ(RTC_CNTL_INT_ST_REG);
    rtc_isr_handler_t* it;
    portENTER_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
    SLIST_FOREACH(it, &s_rtc_isr_handler_list, next) {
        if (it->mask & status) {
            portEXIT_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
            (*it->handler)(it->handler_arg);
            portENTER_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
        }
    }
    portEXIT_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
    REG_WRITE(RTC_CNTL_INT_CLR_REG, status);
}
```
current ble-beta code:
```
static void rtc_isr(void* arg)
{
    uint32_t status = REG_READ(RTC_CNTL_INT_ST_REG);
    rtc_isr_handler_t* it;
    portENTER_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
    SLIST_FOREACH(it, &s_rtc_isr_handler_list, next) {
        if (it->mask & status) {
            portEXIT_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
            (*it->handler)(it->handler_arg);
            portENTER_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
        }
    }
    portEXIT_CRITICAL_ISR(&s_rtc_isr_handler_list_lock);
    REG_WRITE(RTC_CNTL_INT_CLR_REG, status);
}
```
my local code:
```
static void rtc_isr(void* arg)
{
    uint32_t status = REG_READ(RTC_CNTL_INT_ST_REG);
    rtc_isr_handler_t* it;
    portENTER_CRITICAL(&s_rtc_isr_handler_list_lock);
    SLIST_FOREACH(it, &s_rtc_isr_handler_list, next) {
        if (it->mask & status) {
            portEXIT_CRITICAL(&s_rtc_isr_handler_list_lock);
            (*it->handler)(it->handler_arg);
            portENTER_CRITICAL(&s_rtc_isr_handler_list_lock);
        }
    }
    portEXIT_CRITICAL(&s_rtc_isr_handler_list_lock);
    REG_WRITE(RTC_CNTL_INT_CLR_REG, status);
}
```
I dont think i am using `rtc_isr_register` anywhere in my code (vs code cant find any call to it, except esp-idf files).

Like i said, i am using freertos software timer which i can control in runtime:
https://www.freertos.org/FreeRTOS-timers-xTimerCreate.html
when it is active im having crash, when its not active app works good.

mahavirj, 2019-04-11:
@chegewara Your local code seems to be different, (`portENTER_CRITICAL` vs `portENTER_CRITICAL_ISR`) and (`portEXIT_CRITICAL` vs `portEXIT_CRITICAL_ISR`)?

chegewara, 2019-04-11:
@mahavirj Yes, you are correct, sorry im a bit tired today.
I will update my local code and check it. 
Thanks a lot for support.

closed 2019-04-12 by chegewara