espressif/esp-idf #3292: Ble antenna range

hdkim, 2019-04-11:
https://www.espressif.com/en/products/hardware/modules

In the homepage, it say ESP support "+19.5 dBm output at the antenna ensures a good physical range"

is it true?

And does ESP32 support bluetooth class 1?

negativekelvin, 2019-04-11:
The datasheet says max BT transmit power is +12dbm but I think currently it is limited to +7dbm.  Higher power outputs are for wifi tx only. 

Please use esp32.com forum for this type of question.

Weijian-Espressif, 2019-04-13:
@hdkim ESP32's Bluetooth maximum tx power is 7dbm, please refer to `esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level)`

Alvin1Zhang, 2019-05-16:
@hdkim Thanks for reporting the issue. Has your issue been fixed? Thanks.

hdkim, 2019-05-17:
@Alvin1Zhang I am still looking for a way to set tx power to +12dbm, but I have not found it yet

chegewara, 2019-05-17:
> @hdkim ESP32's Bluetooth maximum tx power is 7dbm, please refer to `esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level)`

MAX is +7dBm

negativekelvin, 2019-05-17:
Datasheet should be clarified if +12dbm is not available or applicable to br/edr only

[image]

TianHao-Espressif, 2019-05-20:
For now, ESP32 bluetooth can config max BLE/BR/EDR TX power to +9dbm via APIs.  As some certification requirements, bluetooth should not support so large TX power. So we do +9dbm limitation of it.

Alvin1Zhang, 2019-06-10:
@hdkim Thanks for reporting the issue. Has your issue been resolved / your question been answered? Thanks.

negativekelvin, 2019-06-10:
@TianHao-Espressif are you talking about this https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2019/03/26/using_afh_considerin-emyv ?

What if we don't need to be etsi compliant or we have lossy rf antenna path ?

hdkim, 2019-06-10:
@TianHao-Espressif I don't need certification. So I am still looking for a way to do it.
Now I am considering to use External PA (power amplifier) [SKY66112 ](http://www.skyworksinc.com/uploads/documents/SKY66112_11_203225L.pdf) to make its power be higher.
To support this module, ESP32 should support BT tx/rx status event. but there is no api or event to get status of TX/RX right now.

@Alvin1Zhang Not yet, do you have any plan to support +12dbm power via Api in the future?

jack0c, 2024-02-07:
@hdkim chips after ESP32C3 can support 21dBm power https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/bt/include/esp32c3/include/esp_bt.h#L373

jack0c, 2024-02-07:
@Weijian-Espressif looks should add support of 12dBm for esp32

Weijian-Espressif, 2024-02-07:
@jack0c I confirmed with LXF that the maximum TX power of Bluetooth is supported to 9 dbm.

closed 2024-09-23 by Alvin1Zhang