espressif/esp-idf #3293: cmake: COMPONENT_ADD_INCLUDEDIRS not accepting generator expressions

martijnthe, 2019-04-11:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version: v4.1
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):  v3.1.3
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System: macOS
- Power Supply: USB

## Problem Description

I'm trying to add the build interface of a library to the "main" component, using a [cmake generator expression], like you would normally do when using [`target_include_directories()`]:

```
# mylib.cmake:

add_library(mylib INTERFACE)
target_include_directories(mylib INTERFACE .)

# CMakeLists.txt:
include(path/to/mylib.cmake)

set(COMPONENT_SRCS
                   "cmd_system.c"
                   "cmd_wifi.c"
                   "console_example_main.c")
set(COMPONENT_ADD_INCLUDEDIRS
                   $<BUILD_INTERFACE:$<TARGET_PROPERTY:mylib,INTERFACE_INCLUDE_DIRECTORIES>>
                   ".")
```

[cmake generator expression]: https://cmake.org/cmake/help/v3.13/manual/cmake-generator-expressions.7.html?highlight=generator

[`target_include_directories()`]: https://cmake.org/cmake/help/v3.13/command/target_include_directories.html#command:target_include_directories

### Expected Behavior

I would expect this to work, since using cmake generator expressions is idiomatic cmake.

### Actual Behavior

The build fails with an error:
```
CMake Error at ~/esp32/esp-idf/tools/cmake/components.cmake:85 (message):

  ~/esp32/console/main/CMakeLists.txt:
  COMPONENT_ADD_INCLUDEDIRS entry
  '$<BUILD_INTERFACE:$<TARGET_PROPERTY:mylib,INTERFACE_INCLUDE_DIRECTORIES>>'
  not found
Call Stack (most recent call first):
  main/CMakeLists.txt:11 (register_component)
```

The error is coming from this check in [components.cmake]:
```
        get_filename_component(abs_dir ${include_dir} ABSOLUTE BASE_DIR ${component_dir})
        if(NOT IS_DIRECTORY ${abs_dir})
            message(FATAL_ERROR "${CMAKE_CURRENT_LIST_FILE}: "
                "COMPONENT_PRIV_INCLUDEDIRS entry '${include_dir}' does not exist")
        endif()
```

[components.cmake]: https://github.com/espressif/esp-idf/blob/master/tools/cmake/components.cmake#L90

### Steps to repropduce

1. copy the "system/console" project
2. create a mylib.cmake and add changes as shown above in the problem description.
3. run `idf.py build`

### Code to reproduce this issue

See problem description.

## Debug Logs

```
        get_filename_component(abs_dir ${include_dir} ABSOLUTE BASE_DIR ${component_dir})
        if(NOT IS_DIRECTORY ${abs_dir})
            message(FATAL_ERROR "${CMAKE_CURRENT_LIST_FILE}: "
                "COMPONENT_PRIV_INCLUDEDIRS entry '${include_dir}' does not exist")
        endif()
```

renzbagaporo, 2019-04-24:
It seems the only thing preventing this from working is the check if the values of COMPONENT_ADD_INCLUDEDIRS is a directory or not, so it certainly is possible to have it support generator expressions. In the meantime, you can use `target_include_directories` directly:

```
target_include_directories(${COMPONENT_NAME} PUBLIC $<BUILD_INTERFACE:$<TARGET_PROPERTY:mylib,INTERFACE_INCLUDE_DIRECTORIES>>)
```

martijnthe, 2019-04-24:
Yes, that's what I've been using as a work-around.

closed 2023-08-28 by dobairoland