espressif/esp-idf #3294: WPS not connecting, continually reports SYSTEM_EVENT_STA_DISCONNECTED

crc1021, 2019-04-11:
## Environment

- Development Kit:      none
- Kit version
- Module or chip used:  ESP32-WROOM-32D
- IDF version v3.3-beta2-157-g7bbbf2d4b
- Build System:         CMake
- Compiler version crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description
ESP does not connect with AP when doing WPS.  This works in earlier revisions of the idf such as v3.2.  I had commented about this in issue #3246

### Expected Behavior
ESP should connect to AP

### Actual Behavior
ESP reports many SYSTEM_EVENT_STA_DISCONNECTED and eventually times out.

### Steps to reproduce
run the stock idf example under examples/wifi/wps

## Debug Logs

```
I (272) wifi: Init management short buffer num: 32
I (272) wifi: Init static rx buffer size: 1600
I (272) wifi: Init static rx buffer num: 10
I (282) wifi: Init dynamic rx buffer num: 32
I (382) wifi: mode : sta (30:ae:a4:d4:23:2c)
I (382) example_wps: start wps...
I (382) example_wps: SYSTEM_EVENT_STA_START
I (392) wpa: manufacturer: ESPRESSIF, model number: ESP32, model name: ESPRESSIF IOT, device name: ESP STATION
I (392) wpa: wifi_wps_enable

I (12292) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (14342) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
...
I (117122) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (119182) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (120402) example_wps: SYSTEM_EVENT_STA_WPS_ER_TIMEOUT
I (120402) wpa: wifi_wps_disable

```

TimXia, 2019-05-06:
Please refer https://github.com/espressif/esp-idf/issues/3318

crc1021, 2019-05-06:
it is working for me in master now, but would like to see it in v3.3 branch

wimdewitte, 2019-05-09:
WPS (button mode) is indeed not working anymore!
We are using branch 3.2 (as part of Mongoose-OS).
This is problematic for our product, we cannot release our SW; even not our HW into the market.
Espressif, please fix this.

crc1021, 2019-05-14:
the v3.3 branch is now working for me, specifically v3.3-beta3-26-g31c94b25c

Alvin1Zhang, 2019-05-24:
@crc1021 Glad that the v3.3 branch is now working for you. Feel free to reopen if the issue still exists. Thanks.

closed 2019-05-24 by Alvin1Zhang