espressif/esp-idf #3295: Build fail on v4.0-dev-311(linker script generation failed) when ESP32_WIFI_IRAM_OPT = n

no1seman, 2019-04-11:
## Environment

- Development Kit: ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-311-g70eda3d22
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB|external 5V|Battery

## Problem Description
Project build fails if ESP32_WIFI_IRAM_OPT = n after update to v4.0-dev-311-g70eda3d22 esp-idf

### Expected Behavior
Project must be built

### Actual Behavior
Linker error (see log below)
### Steps to repropduce

1. idf.py build
2. ...

### Code to reproduce this issue
Any example that using wifi

## Debug Logs

Checking Python dependencies...
Python requirements from /home/user/esp/esp-mdf/esp-idf/requirements.txt are satisfied.
Running cmake in directory /home/user/esp/beacon/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /home/user/esp/beacon"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/user/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/user/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building empty aws_iot component due to configuration
-- Component names: xtensa esp_common esp_rom soc log heap app_trace freertos vfs newlib esp_ringbuf driver mbedtls wpa_supplicant efuse bootloader_support partition_table app_update spi_flash nvs_flash esp_wifi ethernet lwip tcpip_adapter esp_event pthread smartconfig_ack espcoredump esp32 cxx libtelnet asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota openssl esp_https_server esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt protobuf-c protocomm spiffs ulp unity wifi_provisioning main
-- Component paths: /home/user/esp/esp-mdf/esp-idf/components/xtensa;/home/user/esp/esp-mdf/esp-idf/components/esp_common;/home/user/esp/esp-mdf/esp-idf/components/esp_rom;/home/user/esp/esp-mdf/esp-idf/components/soc;/home/user/esp/esp-mdf/esp-idf/components/log;/home/user/esp/esp-mdf/esp-idf/components/heap;/home/user/esp/esp-mdf/esp-idf/components/app_trace;/home/user/esp/esp-mdf/esp-idf/components/freertos;/home/user/esp/esp-mdf/esp-idf/components/vfs;/home/user/esp/esp-mdf/esp-idf/components/newlib;/home/user/esp/esp-mdf/esp-idf/components/esp_ringbuf;/home/user/esp/esp-mdf/esp-idf/components/driver;/home/user/esp/esp-mdf/esp-idf/components/mbedtls;/home/user/esp/esp-mdf/esp-idf/components/wpa_supplicant;/home/user/esp/esp-mdf/esp-idf/components/efuse;/home/user/esp/esp-mdf/esp-idf/components/bootloader_support;/home/user/esp/esp-mdf/esp-idf/components/partition_table;/home/user/esp/esp-mdf/esp-idf/components/app_update;/home/user/esp/esp-mdf/esp-idf/components/spi_flash;/home/user/esp/esp-mdf/esp-idf/components/nvs_flash;/home/user/esp/esp-mdf/esp-idf/components/esp_wifi;/home/user/esp/esp-mdf/esp-idf/components/ethernet;/home/user/esp/esp-mdf/esp-idf/components/lwip;/home/user/esp/esp-mdf/esp-idf/components/tcpip_adapter;/home/user/esp/esp-mdf/esp-idf/components/esp_event;/home/user/esp/esp-mdf/esp-idf/components/pthread;/home/user/esp/esp-mdf/esp-idf/components/smartconfig_ack;/home/user/esp/esp-mdf/esp-idf/components/espcoredump;/home/user/esp/esp-mdf/esp-idf/components/esp32;/home/user/esp/esp-mdf/esp-idf/components/cxx;/home/user/esp/beacon/components/libtelnet;/home/user/esp/esp-mdf/esp-idf/components/asio;/home/user/esp/esp-mdf/esp-idf/components/jsmn;/home/user/esp/esp-mdf/esp-idf/components/aws_iot;/home/user/esp/esp-mdf/esp-idf/components/bootloader;/home/user/esp/esp-mdf/esp-idf/components/bt;/home/user/esp/esp-mdf/esp-idf/components/coap;/home/user/esp/esp-mdf/esp-idf/components/console;/home/user/esp/esp-mdf/esp-idf/components/nghttp;/home/user/esp/esp-mdf/esp-idf/components/esp-tls;/home/user/esp/esp-mdf/esp-idf/components/esp_adc_cal;/home/user/esp/esp-mdf/esp-idf/components/tcp_transport;/home/user/esp/esp-mdf/esp-idf/components/esp_http_client;/home/user/esp/esp-mdf/esp-idf/components/esp_http_server;/home/user/esp/esp-mdf/esp-idf/components/esp_https_ota;/home/user/esp/esp-mdf/esp-idf/components/openssl;/home/user/esp/esp-mdf/esp-idf/components/esp_https_server;/home/user/esp/esp-mdf/esp-idf/components/esptool_py;/home/user/esp/esp-mdf/esp-idf/components/expat;/home/user/esp/esp-mdf/esp-idf/components/wear_levelling;/home/user/esp/esp-mdf/esp-idf/components/sdmmc;/home/user/esp/esp-mdf/esp-idf/components/fatfs;/home/user/esp/esp-mdf/esp-idf/components/freemodbus;/home/user/esp/esp-mdf/esp-idf/components/idf_test;/home/user/esp/esp-mdf/esp-idf/components/json;/home/user/esp/esp-mdf/esp-idf/components/libsodium;/home/user/esp/esp-mdf/esp-idf/components/mdns;/home/user/esp/esp-mdf/esp-idf/components/mqtt;/home/user/esp/esp-mdf/esp-idf/components/protobuf-c;/home/user/esp/esp-mdf/esp-idf/components/protocomm;/home/user/esp/esp-mdf/esp-idf/components/spiffs;/home/user/esp/esp-mdf/esp-idf/components/ulp;/home/user/esp/esp-mdf/esp-idf/components/unity;/home/user/esp/esp-mdf/esp-idf/components/wifi_provisioning;/home/user/esp/beacon/main
-- IDF_VER: v4.0-dev-311-g70eda3d22
-- Project version: v.0.0.5-72-g3a00e74
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found Perl: /usr/bin/perl (found version "5.26.1") 
-- Adding linker script /home/user/esp/beacon/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script /home/user/esp/esp-mdf/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Building empty aws_iot component due to configuration
-- Component libraries: 
[... 85 lines cut ...]
[48/58] Building C object esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/secure_boot.c.obj
[49/58] Building C object esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/flash_qio_mode.c.obj
[50/58] Building C object esp-idf/main/CMakeFiles/idf_component_main.dir/bootloader_start.c.obj
[51/58] Linking C static library esp-idf/main/libmain.a
[52/58] Building C object esp-idf/bootloader_support/CMakeFiles/idf_component_bootloader_support.dir/src/bootloader_init.c.obj
[53/58] Generating idf_component_main.sections_info
[54/58] Linking C static library esp-idf/bootloader_support/libbootloader_support.a
[55/58] Generating idf_component_bootloader_support.sections_info
[56/58] Building C object CMakeFiles/bootloader.elf.dir/dummy_main_src.c.obj
[57/58] Linking C executable bootloader.elf
[58/58] Generating ../bootloader.bin
esptool.py v2.7-dev
[1077/1080] Generating esp32.project.ld
FAILED: esp-idf/esp32/esp32.project.ld 
cd /home/user/esp/beacon/build/esp-idf/esp32 && /usr/bin/python /home/user/esp/esp-mdf/esp-idf/tools/ldgen/ldgen.py --config /home/user/esp/beacon/sdkconfig --fragments /home/user/esp/esp-mdf/esp-idf/components/xtensa/linker.lf	/home/user/esp/esp-mdf/esp-idf/components/soc/linker.lf	/home/user/esp/esp-mdf/esp-idf/components/hea [... cut ...]
linker script generation failed for /home/user/esp/esp-mdf/esp-idf/components/esp32/ld/esp32.project.ld.in
**ERROR: failed to parse /home/user/esp/esp-mdf/esp-idf/components/esp_wifi/linker.lf
fragment requires at least 1 values for key 'entries' (at char 156), (line:13, col:1)**
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3069212/sdkconfig.zip)

avanderpluijm, 2019-04-18:
Hi, I have similar behavior: 
IDF: v4.0-dev-402-ga20d02b7f 
compiler: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0 
os: macOS 

``[2/5] Generating esp32.project.ld
FAILED: esp-idf/esp32/esp32.project.ld 
cd /Users/Arn/Documents/iota_cclient_esp32/build/esp-idf/esp32 && /usr/bin/python /Users/Arn/esp/esp-idf/tools/ldgen/ldgen.py --config /Users/Arn/Documents/iota_cclient_esp32/sdkconfig --fragments /Users/Arn/esp/esp-idf/components/soc/linker.lf	/Users/Arn/esp/esp-idf/components/xtensa/linker.lf	/Users/Arn/esp/esp-idf/components/heap/linker.lf	/Users/Arn/esp/esp-idf/components/app_trace/linker.lf	/Users/Arn/esp/esp-idf/components/freertos/linker.lf	/Users/Arn/esp/esp-idf/components/newlib/newlib.lf	/Users/Arn/esp/esp-idf/components/esp_ringbuf/linker.lf	/Users/Arn/esp/esp-idf/components/spi_flash/linker.lf	/Users/Arn/esp/esp-idf/components/esp_wifi/linker.lf	/Users/Arn/esp/esp-idf/components/lwip/linker.lf	/Users/Arn/esp/esp-idf/components/esp_event/linker.lf	/Users/Arn/esp/esp-idf/components/espcoredump/linker.lf	/Users/Arn/esp/esp-idf/components/esp32/linker.lf	/Users/Arn/esp/esp-idf/components/esp32/ld/esp32_fragments.lf --input /Users/Arn/esp/esp-idf/components/esp32/ld/esp32.project.ld.in --output /Users/Arn/Documents/iota_cclient_esp32/build/esp-idf/esp32/esp32.project.ld --sections /Users/Arn/Documents/iota_cclient_esp32/build/ldgen.section_infos --kconfig /Users/Arn/esp/esp-idf/Kconfig --env COMPONENT_KCONFIGS=\ /Users/Arn/esp/esp-idf/components/app_trace/Kconfig\ /Users/Arn/esp/esp-idf/components/bt/Kconfig\ /Users/Arn/esp/esp-idf/components/driver/Kconfig\ /Users/Arn/esp/esp-idf/components/efuse/Kconfig\ /Users/Arn/esp/esp-idf/components/esp32/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_adc_cal/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_event/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_http_client/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_http_server/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_https_ota/Kconfig\ /Users/Arn/esp/esp-idf/components/esp_wifi/Kconfig\ /Users/Arn/esp/esp-idf/components/espcoredump/Kconfig\ /Users/Arn/esp/esp-idf/components/ethernet/Kconfig\ /Users/Arn/esp/esp-idf/components/fatfs/Kconfig\ /Users/Arn/esp/esp-idf/components/freemodbus/Kconfig\ /Users/Arn/esp/esp-idf/components/freertos/Kconfig\ /Users/Arn/esp/esp-idf/components/heap/Kconfig\ /Users/Arn/esp/esp-idf/components/libsodium/Kconfig\ /Users/Arn/esp/esp-idf/components/log/Kconfig\ /Users/Arn/esp/esp-idf/components/lwip/Kconfig\ /Users/Arn/esp/esp-idf/components/mbedtls/Kconfig\ /Users/Arn/esp/esp-idf/components/mdns/Kconfig\ /Users/Arn/esp/esp-idf/components/mqtt/Kconfig\ /Users/Arn/esp/esp-idf/components/nvs_flash/Kconfig\ /Users/Arn/esp/esp-idf/components/openssl/Kconfig\ /Users/Arn/esp/esp-idf/components/pthread/Kconfig\ /Users/Arn/esp/esp-idf/components/spi_flash/Kconfig\ /Users/Arn/esp/esp-idf/components/spiffs/Kconfig\ /Users/Arn/esp/esp-idf/components/tcpip_adapter/Kconfig\ /Users/Arn/esp/esp-idf/components/unity/Kconfig\ /Users/Arn/esp/esp-idf/components/vfs/Kconfig\ /Users/Arn/esp/esp-idf/components/wear_levelling/Kconfig --env COMPONENT_KCONFIGS_PROJBUILD=\ /Users/Arn/esp/esp-idf/components/app_update/Kconfig.projbuild\ /Users/Arn/esp/esp-idf/components/bootloader/Kconfig.projbuild\ /Users/Arn/esp/esp-idf/components/esptool_py/Kconfig.projbuild\ /Users/Arn/Documents/iota_cclient_esp32/main/Kconfig.projbuild\ /Users/Arn/esp/esp-idf/components/partition_table/Kconfig.projbuild --env IDF_CMAKE=y --env IDF_PATH=/Users/Arn/esp/esp-idf --env IDF_TARGET=esp32
linker script generation failed for /Users/Arn/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: failed to parse /Users/Arn/esp/esp-idf/components/esp_event/linker.lf
required keys ['entries'] for fragment not found (at char 38), (line:2, col:5)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1``

tomseago, 2019-04-19:
While I did not test the issue originally reported with the `esp_wifi/linker.lf` file, I did run into the second issue with `esp_events/linker.lf` that appears to be the result of the changing linker generation syntax. I have submitted the following pull request to address the issue in the second file. 

https://github.com/espressif/esp-idf/pull/3335

closed 2019-04-29 by igrr