espressif/esp-idf #3298: ERROR*** A stack overflow in task sc_ack_send_tas has been detected. abort() was called at PC 0x4008eae0 on core 1

filzek, 2019-04-11:
Latest Master as of 04/11/2019

 (75102) wl_flash: config - config result: state_size=0x00001000, cfg_size=0x00001000, addr_cfg=0x000
ea000, addr_state1=0x000e***ERROR*** A stack overflow in task sc_ack_send_tas has been detected.
abort() was called at PC 0x4008eae0 on core 1

It seens that there is a problem now with smartconfig SC_ACK_TASK_STACK_SIZE  it seens to be small or there is a new problem in the sdk.

filzek, 2019-04-11:
In smartconfig_ack.h

need to set it as

#define SC_ACK_TASK_STACK_SIZE           4096

and now the problem is fixed, 

I am debuging the entire firmware so, enable all poisoning and tail/head checks all around, maybe this could cause this problem in task stack size for the ack in smartconfig.

closed 2019-05-22 by filzek

xu18838022837, 2022-05-23:
good