espressif/esp-idf #3299: Illegal folder name ("con") in build folder

freedaun, 2019-04-11:
The exact path created is:

   C:\msys32\home\user\projects\inv660-dev\project\build\include\config\smp\slave\con\params\upd\enable.h

Note this path contains \con\ which is illegal on Windows. The result is that ANY folder-copy that includes the \con\ folder will (silently!) fail. 

This is documented on:
   https://docs.microsoft.com/en-us/windows/desktop/FileIO/naming-a-file 

Don't know the IDF version, but it is very recent, from last week or so, probably a new setting in menuconfig. Copied this from the output:

  Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
  Compiler version: 5.2.0

projectgus, 2019-04-15:
Hi @freedaun ,

Thanks for reporting this. We don't use this kconfig feature of "split" include directories (where each part of the config symbol name becomes part of the directory tree)  at all, but it's enabled by default.

The BT config item which creates this directory was added in d71df18d923f5f6584874039c779b24b3cf6eb74, but the directory is only created if the option is enabled.

A fix for this will be available ASAP.

ghubuser987, 2019-04-17:
"con" is present even in the hello_world example.

Example:
c:\esp32\idf\msys32\home\ann_teo\esp\hello_world\build\include\config\smp\slave\con

This leaves a bad impression.  Unless one already knows that "con" is a reserved name in Windows, it looks as if expressif software corrupted the Windows file system.

Note that from the Windows command prompt, "con" can be deleted with the \\.\ prefix:
rd \\.\c:\esp32\idf\msys32\home\swee_teo\esp\hello_world\build\include\config\smp\slave\con

freedaun, 2019-04-17:
> Note that from the Windows command prompt, "con" can be deleted with the \.\ prefix:

or use WinZip or WinRar and chose 'delete files' after archiving.

projectgus, 2019-04-18:
> "con" is present even in the hello_world example.

I can't reproduce this on Windows unless I [enable this config option](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-smp-slave-con-params-upd-enable).

The directory will stick around after this option is disabled again, though (especially given Windows will refuse to delete it!)

A fix for GNU Make users is in internal review and should be merged in then next few days.

CMake users will have to wait a little bit longer as we ship a precompiled `mconf-idf` binary as part of IDF Tools package (this is built from the kconfig-frontends repo linked above).

> `\ann_teo\`
> `\swee_teo\`

Cute. ;)

fronders, 2019-09-11:
hey there! it's been a long time, now that v3.3 LTS has been released - it still has the issue :(

projectgus, 2019-09-12:
Hi @fronders ,

Yes, sorry the bug has been fixed in master branch (which was pre-v4.0 at the time) and will be fixed in IDF v4.0 release but the backport to v3.3 was only merged yesterday, so this will not be fixed there until v3.3.1 (should be released in a few weeks).

Sorry for the inconvenience.

siqneibi, 2020-04-08:
For anyone that still has this issue, you can delete from the windows command prompt with \\.\ prefix like this:
rd /S /Q \\.\C:\msys32\home\esp\hello_world\include\config\smp\slave\con

dobairoland, 2020-04-28:
This has been fixed in all supported releases. If you still getting this error the directory should be removed manually as suggested above. ESP-IDF no longer creates these directories.

closed 2020-04-28 by dobairoland