espressif/esp-idf #3301: I2C and UART driver blocks indefinitely

FaBjE, 2019-04-12:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Module or chip used: ESP32-WROVER
- IDF version: Master
- Build System: Platform IO

## Problem Description

There is a bug in the I2C and UART driver functions using the "ticks_to_wait" parameter.

In the beginning of the function the "absolute timeout time" is calculated:
portTickType ticks_end = xTaskGetTickCount() + ticks_to_wait;

Half-way the function the "remaing timeout time" is calculated using the following statement:
ticks_to_wait = ticks_end - xTaskGetTickCount();

This causes unexpected behaviour when calling the function with a low timeout. The worst case is when using zero (FreeRTOS "standard" for a quick polling without blocking).

In case of timeout 0, When a os tick event occurs between the "absolute timeout time" and the "remaining timeout" calculation (and this does happen) the xTaskGetTickCount is larger than the ticks_end value.

Causing the unsigned integer calculation to underflow and cause a timeout of 4294967295 os ticks instead of the expected zero.

### Expected Behaviour
Function executes without blocking when using ticks_to_wait = 0

### Actual Behaviour
Function blocks indefinitely

### Steps to reproduce
See example below

### Code to reproduce this issue

    //Poll for received slave data
    uint8_t rxData = 0;

    while(i2c_slave_read_buffer(1, &rxData, 1, 0) > 0)
    {
            printf("RX: %02X\n", rxData);
    }

### Temporarily workaround
Use ticks_to_wait = 1
but this causes the function to block 1 os tick (which is better than indefinitely ;))

### Suggested solution
Check if the xTaskGetTickCount is lower or equal to the ticks_end. Otherwise use 0 as "remaining timeout time"

KonstantinKondrashov, 2019-05-17:
Hi @FaBjE thanks for your report. I have created a fix MR into our internal repo, it will be merged ASAP.

closed 2019-06-12 by igrr

laurentbourassa, 2020-08-11:
This issue is unresolved. Setting ticks_to_wait to 0 blocks the reading of values (it constantly returns 0) from the buffer indefinitely or until a new value as been added to the buffer. Setting ticks_to_wait to 1 will work as expected.