espressif/esp-idf #3302: Secure boot and secure flash

zavovi, 2019-04-12:
Hi, 
please, could you help me with secure boot and secure flash? I tried this: https://docs.espressif.com/projects/esp-idf/en/latest/security/secure-boot.html#secure-boot-reflashable and it was working. But then I enabled NVS encryption and then not work. I tried disable NVS encryption, but still not working. I can download my application into ESP32, but when I start "monitor" then no text received. It seems some issue with bootloader or partitions, but I don't know, where is the problem. Could you help me please? 

Here are my fuses:
```
espefuse.py v2.6
Connecting........____
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 3 R/W (0x3)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 1 R/W (0x1)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 1 R/W (0x1)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 1 R/W (0x1)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 1 R/W (0x1)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 1 R/W (0x1)
BLK1                   Flash encryption key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK2                   Secure boot key
  = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 384 R/W (0x180)
RD_DIS                 Efuse read disablemask                            = 3 R/W (0x3)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 1 R/W (0x1)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 1 R/W (0x1)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 1 R/W (0x1)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address
  = bc:dd:c2:f1:21:f5 (CRC 9e OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 1 R/W (0x1)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1128 R/W (0x4)

Flash voltage (VDD_SDIO) set to 3.3V by efuse.
```

And I have enabled this options in menuconfig:
```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_ENABLED=y
CONFIG_SECURE_BOOTLOADER_ONE_TIME_FLASH=
CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="tcs_boot_secure_key.pem"
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_256BIT=y
CONFIG_SECURE_BOOTLOADER_KEY_ENCODING_192BIT=
CONFIG_SECURE_BOOT_INSECURE=
CONFIG_FLASH_ENCRYPTION_ENABLED=y
CONFIG_FLASH_ENCRYPTION_INSECURE=

CONFIG_PARTITION_TABLE_OFFSET=0x10000

#
# NVS
#
CONFIG_NVS_ENCRYPTION=

```

My partition table:

```
nvs,      data, nvs,     0x11000,  0x6000,
phy_init, data, phy,     0x17000,  0x1000,
factory,  app,  factory, 0x20000, 2M,
```

Thank you

zavovi, 2019-04-12:
One additional thing: I used `make flash-erase` before it stopped working.

negativekelvin, 2019-04-13:
Your FLASH_CRYPT_CNT=3 means encryption should be disabled and you should be able to reflash plaintext signed binaries.  But if you don't have the flash encryption key be very careful, you only get to flash 4 times total and in this case you should leave encryption off during development because you are not using OTA update.  You need to have the flash encryption key if you want to reflash pre-encrypted signed binaries without incrementing the FLASH_CRYPT_CNT.

projectgus, 2019-04-17:
Hi @zavovi ,

As @negativekelvin says, currently there are two bits set in `FLASH_CRYPT_CNT` so encryption should be disabled (even number of bits set). Do you have a copy of your flash encryption key that is written in efuse, as well as the secure boot key?

>  can download my application into ESP32, but when I start "monitor" then no text received. It seems some issue with bootloader or partitions, but I don't know, where is the problem.

When you say "no text received", do you mean literally nothing? There should be some output from the ESP32 whenever it starts up, even if it can't successfully boot. The only time you should get absolutely zero output from a booting ESP32 would be if the GPIO15 (MTDO) pin is pulled low at reset, which suppresses all boot messages.

If GPIO15 is pulled high (or disconnected) then the ESP32 is not starting for some reason, or the startup output is not being captured successfully via serial.

If there is some output, can you please let us know what it is? Thanks.

zavovi, 2019-04-17:
Hi,
thank you, for your replies. It seems, that I make mistake. I don't understand everything and now, it is crypted. I tried flash again via UART, and once, it was ok, and the FLASH_CRYPT_CNT incremented. It seems, that now I should change the ESP32 module on my board. I haven't got saved the key for crypted flash, I have only key for signed app. 

I am thinking, that I will use only signed binary and no encryption :-)

Thank you very much!

projectgus, 2019-04-18:
@zavovi Good to hear. Note that if you want to continue to use this module with secure boot and no encryption, you can still do this - use `espefuse.py burn_efuse FLASH_CRYPT_CNT` to increment the flash_crypt_cnt one more bit, which will disable encryption. Just make sure not to flash a bootloader which will re-enable encryption again.

closed 2019-04-18 by projectgus

ManishMehra001, 2019-09-26:
Hi,

This time before I execute my steps on new board,
I want to verify below steps if these steps are ok to execute or i need to add any missing steps here:

Enable Secure Boot And Flash Encryption
Select "Security features"
Enable for "Enable secure boot in bootloader"
Select "Secure bootloader mode", select "Reflashable".
Enable "Sign binaries during build"
On "Secure boot privite signing key", input your file name, here is "my_secure_boot_signing_key.pem"
Enable "Allow potentially insecure options"
Select "Potentially insecure option",
Select "Leave ROM BASIC Interpreter available on reset"
Select "Allow JTAG Debugging"
Select "Allow app partition length not 64KB aligned"
Select "Leave UART bootloader encryption enabled"
Select "Leave UART bootloader decryption enabled"
Select "Leave UART bootloader flash cache enabled"
Select "Bootloader config"
Select "Bootloader log verbosity"
Select "No output"

Select "Enable flash encryption on boot (READ DOCS FIRST) - Enable usage mode (Development(NOT SECURE))"
Save and Exit menu config

make bootloader

Enable Secure boot on board and Burn secure bootloader key

3.1 Enable Secure boot on board by burn ABS_DONE_0 to 1,
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_efuse ABS_DONE_0

3.2 Burn secure bootloader key (secure-bootloader-key-256.bin),
python $IDF_PATH/components/esptool_py/esptool/espefuse.py --port /dev/ttyS29 burn_key secure_boot ./build/bootloader/secure-bootloader-key-256.bin

Build and Flash following bin files at their address using esp32 flash download tool:
bootloader-reflash-digest_En_0x0.bin at address 0x0
Build Partition table and flash at address 0x8000
Build factory app and flash at address 0x10000
Build main app and flash at address 0x100000
Build ota_data_initial and flash at address 0xd000
Build otadata_086b0e29f7c5570fb7ee208a8c7f8c0f and flash at address 0x400000

Reboot the H/W.

Please have a look on my above steps and if anything is missing here please suggest me.
It would be great help of yours if finally we can work with this approach where we dont need to create flash encryption key.

Additional queries to above steps:

Do I need to burn FLASH_CRYPT_CNT and FLASH_CRYPT_CONFIG ? If yes then what values I need to set ?

And also please suggest, do I need to adjust the flash addresses of my binaries at step 4 and if yes I need to adjust the address then what address values I should use for each of binary at step 4.

Thanks & Best Regards,
Manish