espressif/esp-idf #3303: Addition of Json parsing library- Parson

Kiogora, 2019-04-12:
https://github.com/kgabis/parson

I have tested cjson and parson for json assembly and parsing and I believe parson may be a faster alternative, just based on my tests of the two. It is also very well written.

I am therefore putting this up as a feature addition.

davydnorris, 2019-04-14:
I'm using nx_json as a parser. It's really good and parses in place so very low memory usage.

X-Ryl669, 2019-04-17:
There's also https://github.com/X-Ryl669/JSON that does not allocate any memory, supports partial parsing (that it, you can start parsing in a HTTP or MQTT client receive's callback, and continue when you have more data available even if the initial buffer was reused). I believe it's as fast as it can be, and only consume 1.65kB of flash size when built on ESP32.

projectgus, 2019-11-25:
Hi @Kiogora ,

I'm sorry noone got back to you when you proposed this in April. I think I must have composed a reply in my head and then never submitted it.

As can be seen in the other comments, there are a number of low level JSON parsing libraries with different relative advantages. What we really want is to make it easy for users to add whatever library they want to their ESP-IDF project.

However, we can't do this by adding every possible JSON library to ESP-IDF : it's a lot of overhead for us to maintain them, check for security fixes, etc. So we'll probably stick with just supporting one - cJSON - in ESP-IDF for now. It may not be the best in every regard, but we already have it.

The good news for incorporating embedded JSON libraries into IDF projects is they often have very few dependencies so it can be easy to convert them to an ESP-IDF component. The [new CMake-based build system](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#using-third-party-cmake-projects-with-components) even makes it possible to re-use an existing library's CMakeLists.txt files, and parson supports CMake already.

Medium- to long-term, we have an ESP-IDF Package Manager in the works which will make it even easier to package third party libraries as ESP-IDF components and then incorporate them into ESP-IDF projects. However this is still in development for now.

Kiogora, 2019-12-07:
Hi @projectgus 

Thank you for the detailed explanation and taking the time to do so. I understand the maintenance overhead.

Thanks also for the tidbits on what you have lined up at Espressif for the medium term future. The openness has made me a solid customer since v2.1 esp-idf.

closed 2019-12-07 by Kiogora