espressif/esp-idf #3305: Build system: multiple rules error

huming2207, 2019-04-13:
## Environment

- Development Kit:      N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROVER-I
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-311-g70eda3d22
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a
- Operating System: Linux (Manjaro Deepin 18)
- Power Supply: 3.3v

## Problem Description

Hi there,

I'm trying to follow the instruction to import `libical` to ESP-IDF CMake project. I'm folllowing this sample project: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_lib

Here is my test project with libical: https://github.com/huming2207/esp-ical-test

Then my CLion passes indexing with no issues. But when I run `idf.py build`, it throws me an error saying `ninja: error: build.ninja:14271: multiple rules generate lib/libical.a [-w dupbuild=err]
`.

I will paste the full output below. Are there any typos in my CMakeList, or bugs in the ESP-IDF build system? 

Regards,
Jackson

### Expected Behavior

The test project should build with no issue

### Actual Behavior

Ninja throws me a `multiple rules` error.

### Steps to repropduce

1. Clone my demo project: https://github.com/huming2207/esp-ical-test
2. Run `idf.py build`

### Code to reproduce this issue

Please refer to my demo project: https://github.com/huming2207/esp-ical-test

## Debug Logs

After run `idf.py build`:

```
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /home/hu/esp/esp-idf/requirements.txt are satisfied.
Running cmake in directory /home/hu/Projects/test-ical/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized /home/hu/Projects/test-ical"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.21.0") 
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
[... 357 lines cut ...]
 * Berkeley DB storage support, build in support for Berkeley DB storage
 * Option ICAL_ERRORS_ARE_FATAL, icalerror_* calls will abort instead of internally signaling an error
 * Option ICAL_ALLOW_EMPTY_PROPERTIES, prevents empty properties from being replaced with X-LIC-ERROR properties
 * Option USE_BUILTIN_TZDATA, use our own timezone data rather then the system timezone data
 * Option GOBJECT_INTROSPECTION, build GObject introspection "typelib" files
 * Option ICAL_GLIB_VAPI, build Vala "vapi" files

-- The following OPTIONAL packages have not been found:

 * BDB, Berkeley DB storage, <http://www.oracle.com/database/berkeley-db>
   For Berkeley DB storage support
 * Doxygen, API Documentation system, <http://www.doxygen.org>
   Needed to build the API documention.

-- Component libraries: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hu/Projects/test-ical/build
Running ninja in directory /home/hu/Projects/test-ical/build
Executing "ninja all"...
ninja: error: build.ninja:14271: multiple rules generate lib/libical.a [-w dupbuild=err]

ninja failed with exit code 1

```

renzbagaporo, 2019-05-22:
Hi @huming2207. Sorry for taking a long time getting back to this issue. Taking a look at the `ical` CMake files, it seems there is an opportunity for multiple rule definitions. For example:

https://github.com/libical/libical/blob/621488e59945ec1e7c1b6fae78ebbe4e3a90eec1/src/libical/CMakeLists.txt#L259
```
add_library(ical ${LIBRARY_TYPE} ${ical_LIB_SRCS})
```
This creates libical.a. However, later in the CMake file:

https://github.com/libical/libical/blob/621488e59945ec1e7c1b6fae78ebbe4e3a90eec1/src/libical/CMakeLists.txt#L262 and https://github.com/libical/libical/blob/621488e59945ec1e7c1b6fae78ebbe4e3a90eec1/src/libical/CMakeLists.txt#L286

```
add_library(ical-static STATIC ${ical_LIB_SRCS})
...
set_target_properties(ical-static PROPERTIES OUTPUT_NAME "ical")
```
This also creates a file with name libical.a, thus resulting to multiple rule error. I suggest reviewing how to configure the library properly.

huming2207, 2019-05-22:
Hi @renzbagaporo 

I've tried setting `SHARED_ONLY` to ON with latest GCC (8.2 2019r1-rc2) and IDF (v4.0-dev-584-g3bf56cdd1), but it shows a lot of these kinds of errors:

```
idf.py build       
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from /home/hu/esp/esp-idf/requirements.txt are satisfied.
Running ninja in directory /home/hu/Projects/esp-ical-test/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster builds
-- Project version: 724c642
-- Building ESP-IDF components for target esp32
CMake Warning at /home/hu/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  version 8.2.0 is not the supported version 5.2.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /home/hu/esp/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  /home/hu/esp/esp-idf/tools/cmake/build.cmake:281 (include)
  /home/hu/esp/esp-idf/tools/cmake/build.cmake:442 (__build_process_project_includes)
  /home/hu/esp/esp-idf/tools/cmake/project.cmake:311 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at /home/hu/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:25 (message):
  Xtensa toolchain /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  does not appear to be built with crosstool-ng.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /home/hu/esp/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  /home/hu/esp/esp-idf/tools/cmake/build.cmake:281 (include)
  /home/hu/esp/esp-idf/tools/cmake/build.cmake:442 (__build_process_project_includes)
  /home/hu/esp/esp-idf/tools/cmake/project.cmake:311 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Adding linker script /home/hu/Projects/esp-ical-test/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/hu/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/hu/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
CMake Warning (dev) at main/lib/ical/cmake/modules/FindICU.cmake:74 (set):
  implicitly converting 'DOC' to 'STRING' type.
Call Stack (most recent call first):
  main/lib/ical/CMakeLists.txt:173 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found ICU version 64.2
-- Found ICU header files in /usr/include
-- Found ICU libraries: /lib64/libicuuc.so /lib/libicui18n.so
-- Optional package Berkeley DB was not found
-- Could NOT find Threads (missing: Threads_FOUND) 
CMake Warning (dev) at main/lib/ical/src/libical/CMakeLists.txt:259 (add_library):
  ADD_LIBRARY called with SHARED option but the target platform does not
  support dynamic linking.  Building a STATIC library instead.  This may lead
  to problems.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at main/lib/ical/src/libical/CMakeLists.txt:328 (add_library):
  ADD_LIBRARY called with SHARED option but the target platform does not
  support dynamic linking.  Building a STATIC library instead.  This may lead
  to problems.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at main/lib/ical/src/libicalss/CMakeLists.txt:80 (add_library):
  ADD_LIBRARY called with SHARED option but the target platform does not
[... 89 lines cut ...]
set(SHARED_ONLY ON CACHE BOOL "build shared libraries only" FORCE)
set(STATIC_ONLY OFF CACHE BOOL "build static libraries only" FORCE)

# Unfortunately the library performs install and export. Would
# have been nice if devs made that an option like BUILD_SHARED_LIBS
# and BUILD_TESTING. Override install() and export() to do nothing
# instead.
function(install)
endfunction()

function(export)
endfunction()

# Import tinyxml2 targets
add_subdirectory(lib/ical)

# Link tinyxml2 to main component
target_link_libraries(${COMPONENT_LIB} ical)

```

Is there any way to fix that?

Regards,
Jackson

huming2207, 2019-05-22:
I've just done a quick google and found this: https://stackoverflow.com/questions/17945266/wrong-cmake-binary-dir

Is it possible that the CMakeList files got copied to `build` directory and screwed up something?

renzbagaporo, 2019-05-22:
Before even that, this looks the library that has dependencies on host stuff (its looking to include headers from `/usr/include` and linking in `/lib64/libicuuc.so` and `/lib/libicui18n.so`, which is definitely not possible). You may have to expend some effort porting the original library and its dependencies.

huming2207, 2019-05-22:
Hi @renzbagaporo , yea indeed, thanks for noticing...I have tried disabled libicu support that but it didn't behave as I expected. So far I will give up playing with this library and maybe write my own ical parser later on

renzbagaporo, 2019-05-22:
Yeah, the example imported lib works because it doesn't need much outside the C/C++ standard library. By the way, if you have no more follow-up questions, mind closing this issue?

huming2207, 2019-05-22:
Yep. Thanks for your help.

closed 2019-05-22 by huming2207