espressif/esp-idf #3307: Build system: Failed to build example idf_as_lib on OSX

thepian, 2019-04-14:
## Environment

- IDF version: v4.0-dev-311-g70eda3d22
- Build System: CMake
- Compiler version: crosstool-ng-1.22.0-80-g6c4433a 5.2.0
- Operating System:   macOS

## Problem Description

I tried to build the SDK example idf_as_lib

### Expected Behavior

Pass.

### Actual Behavior

pyparsing.ParseException: File /Volumes/Projects/esp-idf/examples/build_system/cmake/idf_as_lib/build/esp-idf/xtensa/idf_component_xtensa.sections_info is not a valid sections info file.  (at char 0), (line:1, col:1)
make[2]: *** [esp-idf/esp32/esp32.project.ld] Error 1
make[1]: *** [esp-idf/esp32/CMakeFiles/ldgen_esp32.project.ld_script.dir/all] Error 2
make: *** [all] Error 2

### Steps to repropduce

1. set IDF_PATH & PATH
2. cd idf_as_lib
3. ./build.sh

## Debug Logs

```
Scanning dependencies of target idf_component_pthread
[ 94%] Building C object esp-idf/pthread/CMakeFiles/idf_component_pthread.dir/pthread.c.obj
[ 95%] Building C object esp-idf/pthread/CMakeFiles/idf_component_pthread.dir/pthread_cond_var.c.obj
[ 95%] Building C object esp-idf/pthread/CMakeFiles/idf_component_pthread.dir/pthread_local_storage.c.obj
[ 95%] Linking C static library libpthread.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: libpthread.a the table of contents is empty (no object file members in the library define global symbols)
[ 95%] Built target idf_component_pthread
[ 95%] Generating idf_component_pthread.sections_info
[ 95%] Built target idf_component_pthread_sections_info
Scanning dependencies of target idf_component_smartconfig_ack
[ 95%] Building C object esp-idf/smartconfig_ack/CMakeFiles/idf_component_smartconfig_ack.dir/smartconfig_ack.c.obj
[ 96%] Linking C static library libsmartconfig_ack.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: libsmartconfig_ack.a the table of contents is empty (no object file members in the library define global symbols)
[ 96%] Built target idf_component_smartconfig_ack
[ 96%] Generating idf_component_smartconfig_ack.sections_info
[ 96%] Built target idf_component_smartconfig_ack_sections_info
Scanning dependencies of target idf_component_espcoredump
[ 96%] Building C object esp-idf/espcoredump/CMakeFiles/idf_component_espcoredump.dir/src/core_dump_common.c.obj
[ 96%] Building C object esp-idf/espcoredump/CMakeFiles/idf_component_espcoredump.dir/src/core_dump_flash.c.obj
[ 97%] Building C object esp-idf/espcoredump/CMakeFiles/idf_component_espcoredump.dir/src/core_dump_port.c.obj
[ 97%] Building C object esp-idf/espcoredump/CMakeFiles/idf_component_espcoredump.dir/src/core_dump_uart.c.obj
[ 97%] Linking C static library libespcoredump.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: libespcoredump.a the table of contents is empty (no object file members in the library define global symbols)
[ 97%] Built target idf_component_espcoredump
[ 97%] Generating idf_component_espcoredump.sections_info
[ 97%] Built target idf_component_espcoredump_sections_info
[ 97%] Built target ldgen_section_infos
[ 97%] Generating esp32.project.ld
Traceback (most recent call last):
  File "/Volumes/Projects/esp-idf/tools/ldgen/ldgen.py", line 119, in <module>
    main()
  File "/Volumes/Projects/esp-idf/tools/ldgen/ldgen.py", line 87, in main
    sections_infos.add_sections_info(sections_info_file_obj)
  File "/Volumes/Projects/esp-idf/tools/ldgen/generation.py", line 590, in add_sections_info
    raise ParseException("File " + sections_info_file.name + " is not a valid sections info file. " + p.message)
pyparsing.ParseException: File /Volumes/Projects/esp-idf/examples/build_system/cmake/idf_as_lib/build/esp-idf/xtensa/idf_component_xtensa.sections_info is not a valid sections info file.  (at char 0), (line:1, col:1)
make[2]: *** [esp-idf/esp32/esp32.project.ld] Error 1
make[1]: *** [esp-idf/esp32/CMakeFiles/ldgen_esp32.project.ld_script.dir/all] Error 2
make: *** [all] Error 2
```

thepian, 2019-04-14:
Actually this went away after deleting the build directory. Perhaps there is some room for cleanup, as the only other build I had tried was against v3.3

renzbagaporo, 2019-04-24:
Is it possible for you to retrieve `/Volumes/Projects/esp-idf/examples/build_system/cmake/idf_as_lib/build/esp-idf/xtensa/idf_component_xtensa.sections_info` from deletion?

Alvin1Zhang, 2019-06-05:
@thepian Thanks for reporting the issue, would you help share more logs or details? Thanks.

renzbagaporo, 2019-08-06:
Closed as there is no update and build process no longer generates .sections_info files. @thepian

closed 2019-08-06 by renzbagaporo