espressif/esp-idf #3311: Spiffs File Corruption

sebirdman, 2019-04-15:
## Environment

- Development Kit:     Wrover
- Module or chip used: ESP32-WROVER
- IDF version: f49e2afe946f414e1a9f2d7426af6665eab3731b
- Build System:  make
- Compiler version: 
- Operating System: Linux
- Power Supply: external 5V

## Problem Description

I've received three devices back from customers where files on our spiffs partitions partitions are corrupted. 

My spiffs has a few certs to connect to AWS, as well as a file that holds local logs that happen while the device is offline. Spiffs is only opened once, at boot. My certs are only opened with read permissions, and the only file on the spiffs that has write permissions is the local logging. 

Here is some information that i retrieved from the most recent device:
```

0000 id//dd/did/iiii	era_cnt: 27811
0001 //dddd/iiiiiid/	era_cnt: 27823
0002 d//dddddd/ddiii	era_cnt: 27822
0003 idi____________	era_cnt: 27821
0004 /dddd/d/ddid/ii	era_cnt: 27818
0005 //d/d/diiiiiid/	era_cnt: 27817
0006 //dddddd/iiiii/	era_cnt: 27812
0007 /ddd/iiiiiiiid/	era_cnt: 27813
0008 _______________	era_cnt: 27824
0009 /dddd/dd/iiiii/	era_cnt: 27810
000a /ddddd//ddid///	era_cnt: 27820
000b //d/iiiiiiiid//	era_cnt: 27819
000c _______________	era_cnt: 27825
000d /ddd/dd/d/iiiii	era_cnt: 27806
000e /ddddd/iiiiiiii	era_cnt: 27815
000f //ddd/d/dd/iiii	era_cnt: 27809
0010 _______________	era_cnt: 27826
0011 //d/dd/iiiiiiid	era_cnt: 27816
0012 //ddd/iiiiiiii/	era_cnt: 27814
0013 id//dddd/iiiiii	era_cnt: 27808
0014 ii//dd//iiiiiii	era_cnt: 27807
era_cnt_max: 27827
last_errno:  0
blocks:      21
free_blocks: 3
page_alloc:  190
page_delet:  68
used:        47690 of 71786
total: 71786
used: 47690
```

Here are my spiffs sdkconfig settings:

```
#
# SPI Flash driver
#
CONFIG_SPI_FLASH_VERIFY_WRITE=
CONFIG_SPI_FLASH_ENABLE_COUNTERS=
CONFIG_SPI_FLASH_ROM_DRIVER_PATCH=y
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS=
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED=

#
# SPIFFS Configuration
#
CONFIG_SPIFFS_MAX_PARTITIONS=1

#
# SPIFFS Cache Configuration
#
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=

#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

```

If there is any other information i can provide to help track this down, please let me know.

sebirdman, 2019-04-15:
It appears that the esp_spiffs component does not run spiffs check ever:

https://github.com/pellepl/spiffs/wiki/FAQ#powerlosses-contd-when-should-i-run-spiffs_check

I manually edited the spiffs register component to call this check and recieved the following output on a bad device: 

```
E (278) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 7e:0
E (308) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 92:0
E (318) SPIFFS: CHECK: type:LOOKUP, report:FIX LOOKUP, 1:0
E (318) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, e7:0
E (318) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, e8:0
E (538) SPIFFS: CHECK: type:LOOKUP, report:FIX LOOKUP, 0:0
E (538) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, f9:0
E (558) SPIFFS: CHECK: type:LOOKUP, report:FIX LOOKUP, 0:0
E (558) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, fe:0
E (558) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 118:0
E (558) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 119:0
E (568) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 127:0
E (578) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 12d:0
E (588) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 14d:0
E (588) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 14e:0
E (598) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 11:800e
E (598) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 1e:8044
E (608) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 1f:805c
E (618) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 2c:803d
E (618) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 2d:800a
E (628) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 2e:804d
E (638) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 3d:8007
E (638) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 3e:801b
E (648) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 3f:8047
E (658) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 4c:8040
E (658) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 4d:8030
E (668) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 4e:8039
E (668) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 4f:802e
E (678) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 58:805a
E (688) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 59:8058
E (688) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5a:800f
E (698) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5b:8039
E (708) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5c:8027
E (708) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5d:8009
E (718) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5e:801d
E (728) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 5f:803f
E (728) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 81:802f
E (738) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 82:800c
E (748) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8a:8053
E (748) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8b:8052
E (758) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8c:8037
E (768) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8d:804a
E (768) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8e:802c
E (778) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 8f:8028
E (788) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 91:804c
E (788) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 93:803b
E (798) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 9c:803e
E (808) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 9d:803b
E (808) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 9e:8015
E (818) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, a1:805d
E (818) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, a4:805e
E (828) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, ac:8059
E (838) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, ad:8048
E (838) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, ae:8054
E (848) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, af:8010
E (858) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, b2:803c
E (858) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, b3:8024
E (868) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, b4:8056
E (878) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, bb:8041
E (878) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, bc:800d
E (888) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, bd:8008
E (898) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, be:801a
E (898) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, bf:8025
[... 61 lines cut ...]
E (1448) SPIFFS: CHECK: type:PAGE, report:DELETE PAGE, df:0
E (1458) SPIFFS: CHECK: type:PAGE, report:DELETE PAGE, f8:0
E (1468) SPIFFS: CHECK: type:PAGE, report:DELETE PAGE, 101:0
E (1468) SPIFFS: CHECK: type:PAGE, report:DELETE PAGE, 10a:0
E (1478) SPIFFS: Ran check
```

On a good device without any corruption:

```

E (369) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 71:0
E (379) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 91:0
E (389) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 93:0
E (389) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 94:0
E (399) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 95:0
E (399) SPIFFS: CHECK: type:LOOKUP, report:FIX INDEX, b:9
E (399) SPIFFS: CHECK: type:LOOKUP, report:DELETE PAGE, 96:0
E (429) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 61:8006
E (429) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, d1:8007
E (429) SPIFFS: CHECK: type:INDEX, report:DELETE PAGE, 131:8008
E (479) SPIFFS: Ran check
```

It seems like, my device should have been running check far more often. Unfortunately the bad device was not able to recover the bad files.

Alvin1Zhang, 2019-04-19:
@sebirdman Hi, thanks for reporting the issue, we will look into this soon.

sebirdman, 2019-06-06:
@Alvin1Zhang any update on this?