espressif/esp-idf #3313: esp_mesh_waive_root

Robokishan, 2019-04-16:
so i was working on mesh 
steps what i have done on mesh
i put two esp32 far from router to seprate room from router and then mesh formed.
now i put third esp32 to near the router. as expected it joined the mesh.
now the problem starts.
i called esp_mesh_waive_root(NULL,MESH_VOTE_REASON_ROOT_INITIATED);  and the esp32 near the router got the vote 1.00 now the expected thing is it should become the root but the problem is the previous root stays the same. and the esp32 near the router became node. how is that possible ?
is there anything to do ??
for this thing to work. ?  and i am using esp32 mdf mwifi example

Alvin1Zhang, 2019-04-18:
@Robokishan Hi, can you check if the original root receives "MESH_EVENT_ROOT_SWITCH_REQ" ? and check if the third esp32 receives "MESH_EVENT_ROOT_SWITCH_ACK"? Or u can post ur log here, we'll help analyze what was happening there. Thanks.

Robokishan, 2019-05-13:
ok please give me sometime i am out of station right now.

Alvin1Zhang, 2020-03-04:
@Robokishan Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2020-03-12:
@Robokishan Thanks for reporting, will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-03-12 by Alvin1Zhang