espressif/esp-idf #3314: bt_discovery example

ludo-d, 2019-04-16:
the bt_discovery example doesn't work after running menuconfig and select BR/EDR only in the bleutooth section 
_GENCONFIG

*** Error during update of the configuration._

ps: option listed in README are not valid anymore
Options choose step:
    1. make menuconfig.
    2. enter menuconfig "Component config", choose "Bluetooth"
    3. enter menu Bluetooth, choose "Classic Bluetooth" and do not choose "Release DRAM from Classic BT controller"
    4. choose your options.

blueMoodBHD, 2019-04-25:
@ludo-d option listed in README are not valid anymore, I will update it later. 
But I enter menuconfig `Component config`, choose `Bluetooth`, choose `Bluetooth controller`, set `Bluetooth controller mode` to `BR/EDR only`. Then it works OK. The version I use is the latest master. What is the version you use?

Alvin1Zhang, 2019-05-16:
@ludo-d Thanks for reporting the issue. Could you help share which version you are using? Thanks.

Alvin1Zhang, 2020-10-12:
Thanks for reporting, will close due to short of updates, feel free to reopen with more details, thanks.

closed 2020-10-12 by Alvin1Zhang