espressif/esp-idf #3315: BLE Server: Characteristic with long value truncated to 20 on notify/indicate

mws-rmain, 2019-04-16:
I'm working with the latest commit (70eda3d). I've modified the BLE gatt_server example, adding a task to periodically send notify / indicate (when enabled). I update the attribute with a changing string, 36 characters long. When I enable notify/indicate with 'nRF Connect', the updates occur, but are always truncated to the first 20 characters (i.e., only a single packet is sent, with MTU of 20).

The gatt_server code does make the call 'esp_gatt_set_local_mtu(500)', so it's clear 'nRF Connect' is not requesting a larger MTU. However, shouldn't the BT stack split the long attribute into two packets?

I believe this is the same issue referenced in #1434. Reading logs @chegewara posted, it looks to me like they might be server-side updates as opposed to client side attribute writes.

For reference, I'm implmenting a BLE virtual UART (loosely paterned after Nordic), with kai morich's excellent android [Serial Bluetooth Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_CA) as my android testbench.

chegewara, 2019-04-16:
Hi,
its up to you if value will be split before sending notification. 
Like you said, nRF connect does not request MTU change and your setup is using MTU = 23, but you can change MTU from nRF Connect after connecting to esp32. Its option o/n top right on connected device screen.

mws-rmain, 2019-04-17:
Thanks, @chegewara. Changing the MTU size will be beneficial, but I'd prefer the BLE stack take care of splitting attribute update packets where the size of the attribute exceeds the MTU. Kai Morich confirmed that 'Bluetooth LE 4.2 DLE (MTU above 20) is not supported' in his Serial Bluetooth Terminal app. So if I do go this route, I won't be able to use his app for testing, and if the ESP-IDF BT stack could split long attribute updates, that would address my througput requirements.

I did try manually splitting the attribute and a loop to perform the update, but througput poor, and the updates quickly lost integrity (packets were lost). I think these issues are partly related to the semaphore handling in the ESP32Snippets BLE wrappers. One specific issue with them is the dispatch of EVERY event to handlers for every service/characteristic/attribute instead of intelligently using the handle (available in many events) to send the event ONLY to the appropriate handler. More intelligent handling of events will be important if handling of long attributes is addressed (it will complicate semaphore handling). I'll post to ESP32Snippets regarding this.

It's clear from issue #1434 that you were involved in that client-side writes are automatically split & the packets queued for a maximum-throughput, low-latency transfer, so the code to do this exists in the stack. I see no reason the same procedure should not apply to server-side updates (i.e, writes to the client) shouldn't benefit from the same procedure. While investigating, I found a [reference](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2015/08/06/_reference_ble_mas-gviy) that (if I'm interpreting correctly) says the SiLabs BT stack transparently handles large attribute splitting & queuing & indicate/notify. I understand that indicate/notify occurs automatically (when enabled) simply by invoking their 'attributes_write' command to update the value of the attribute (server-side).

I've posted my modified gatts_demo.c in this [gist](https://gist.github.com/mws-rmain/6a52bb42fda92617e8c2d280aadf9c3c). The most significant changes are:
- receipt of indicate/notify enables just set a flag (which is actually more realistic/usefull than the existing behavior of this demo);
- a task is created that periodically (100ms) sends an indicate/notify to any enabled client. The attribute value is the string 'abcdefghijklmnopqrstuvwxyz0123456789', but scrolls by one character for each update.

mws-rmain, 2019-04-17:
I ran a test usiing nRFConnect to request a larger MTU (thanks @chegewara), and saw expected results (full string transfer, no truncation errors in the ESP log). I have also requested that Kai Morich add a 'request larger MTU' setting in his app, which will allow me to continue using his app for testing.

The question remains: is this a 'bug' (which should/will be fixed), or a 'limitation', which will just be documented & left). I'd suggest that automatic packet splitting & queuing already occurs on client-side writes, so it would make sense to make server-side updates consistent. Not to mention at least SiLabs appears to already handle this.

chegewara, 2019-04-17:
This is "limitation" that is described in bluetooth specs.

If you need/want splitting value during sending notifications/indications it should be done in your app, what is implemented in esp-idf ble stack its splitting characteristic value when client is trying to read characteristic and value length is bigger than MTU up to 600 bytes length (this is also described in bluetooth specs).

mws-rmain, 2019-04-22:
For posterity, I reviewed the Bluetooth specification (v4.2 & v5.1), and found the sections relevant to Notification (4.10), Indication (4.11) and write long characteristic values (4.9.4). It does appear (as @chegewara states) that notification & indication of new characteristic values is limited by the current MTU.

After re-reading the [SiLabs BT stack notes](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2015/08/06/_reference_ble_mas-gviy), it becomes clear that the information presented is confused: at one point indicating that a (server-side) write triggers notify/indicate, but is size-limited to 255 bytes data, 64 bytes updated at a time, and later confirming that notify/indicate events (client-side) are limted to 20 byte payload.

I am proceding to update my code to increase the MTU, and have requested support for larger MTU in a future release of the Bluetooth Serial Terminal App.

closed 2019-04-22 by mws-rmain