espressif/esp-idf #3318: WPS WiFi provisioning receives encrypted ap, but empty password leading to SYSTEM_EVENT_STA_DISCONNECTED loop

cduvenhorst, 2019-04-17:
When running the wps example, the WPS PBC WiFi provisioning receives encrypted ap, but an empty password. This seems to lead to a SYSTEM_EVENT_STA_DISCONNECTED loop.

WPS PBC WiFi provisioning seems to be broken at least for the  AVM Fritzbox 7490 router.

I (29731) wifi: Start wifi scan
I (29732) wifi: filter: set rx policy=3
I (31784) wifi: filter: set rx policy=4
I (31885) wifi: Start wifi scan
I (31885) wifi: filter: set rx policy=3
I (33937) wifi: filter: set rx policy=4
I (33938) wifi: Start wifi disconnect
I (33938) wifi: connect step 0 -> 0, status=0, err_time=0
I (33939) wifi: connect status 0 -> 0, step=0, err_time=0
I (33944) wifi: filter: set rx policy=8
I (33947) wifi: clear blacklist
I (33950) wifi: Start wifi connect
I (33953) wifi: connect step 0 -> 0, status=0, err_time=0
I (33958) wifi: connect chan=0
I (33961) wifi: connect status 0 -> 1, step=0, err_time=0
I (33966) wifi: connect step 0 -> 1, status=1, err_time=0
I (33971) wifi: filter: set rx policy=3
I (34588) wifi: receive encrypted ap, but password is empty
I (34660) wifi: receive encrypted ap, but password is empty
I (36027) wifi: filter: set rx policy=4
I (36027) wifi: handoff_cb: status=0
I (36028) wifi: clear rc list
I (36028) wifi: clear blacklist
I (36029) wifi: send disconnect event
I (36032) wifi: connect step 1 -> 3, status=1, err_time=1
I (36037) wifi: disable connect timer
I (36041) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (36046) wifi: Start wifi connect
I (36049) wifi: connect step 3 -> 0, status=1, err_time=1
I (36054) wifi: connect chan=0
I (36057) wifi: connect status 1 -> 1, step=0, err_time=0
I (36062) wifi: connect step 0 -> 1, status=1, err_time=0
I (36068) wifi: filter: set rx policy=3
I (36578) wifi: receive encrypted ap, but password is empty
I (36689) wifi: receive encrypted ap, but password is empty
I (36709) wifi: receive encrypted ap, but password is empty
I (36815) wifi: receive encrypted ap, but password is empty
I (38123) wifi: filter: set rx policy=4
I (38124) wifi: handoff_cb: status=0
I (38124) wifi: clear rc list
I (38124) wifi: clear blacklist
I (38125) wifi: send disconnect event
I (38129) wifi: connect step 1 -> 3, status=1, err_time=1
I (38134) wifi: disable connect timer
I (38137) example_wps: SYSTEM_EVENT_STA_DISCONNECTED
I (38143) wifi: Start wifi connect
I (38146) wifi: connect step 3 -> 0, status=1, err_time=1
I (38151) wifi: connect chan=0
I (38154) wifi: connect status 1 -> 1, step=0, err_time=0
I (38159) wifi: connect step 0 -> 1, status=1, err_time=0
I (38164) wifi: filter: set rx policy=3
I (38782) wifi: receive encrypted ap, but password is empty
I (38859) wifi: receive encrypted ap, but password is empty
I (40219) wifi: filter: set rx policy=4
I (40220) wifi: handoff_cb: status=0
I (40220) wifi: clear rc list
I (40220) wifi: clear blacklist
I (40221) wifi: send disconnect event
I (40225) wifi: connect step 1 -> 3, status=1, err_time=1
I (40230) wifi: disable connect timer
I (40233) example_wps: SYSTEM_EVENT_STA_DISCONNECTED

TimXia, 2019-04-19:
@cduvenhorst We have found the root cause and will fix it soon.

TimXia, 2019-05-06:
@cduvenhorst The bug has been fixed on master branch. Please try again.

cduvenhorst, 2019-05-10:
Works like charm. Fixed.

closed 2019-05-10 by cduvenhorst