espressif/esp-idf #3320: Replace to strcmp from strncmp on examples/system/console/cmd_nvs.c

QinYUN575, 2019-04-17:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version : v3.3-beta2-18-g0f927791b
- Build System: Make
- Compiler version : (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Ubuntu 18.04 bionic [Ubuntu on Windows 10]
- Power Supply: USB

Thanks for useful example projects.

----

## Problem Description

when using nvs_set <key> <type> -v <value>
<type> = i32343 can also be executed normally. Look at the code and find that strncmp is used instead of strcmp, which causes the same prefix to be executed.

### Expected Behavior

### Actual Behavior

```log
esp32> nvs_set Var1 i32 -v 213
I (3044434) cmd_nvs: Value stored under key 'Var1'
esp32> nvs_set Var1 i3223 -v 213
I (3050794) cmd_nvs: Value stored under key 'Var1'
esp32>
```

### Code to reproduce this issue

> examples/system/console/cmd_nvs.c

```cpp
static nvs_type_t str_to_type(const char *type)
{
    for (int i = 0; i < TYPE_STR_PAIR_SIZE; i++) {
        const type_str_pair_t *p = &type_str_pair[i];
        if (strncmp(type, p->str, strlen(p->str)) == 0) {
            return  p->type;
        }
    }

    return NVS_TYPE_ANY;
}
```

shivani-tipnis, 2019-05-02:
@QinYUN575 Thank you for pointing this out. The fix is added. Should be available soon.

shivani-tipnis, 2019-05-03:
@QinYUN575 Thank you for pointing this out. The fix has been added and merged. Should be available soon.

closed 2019-05-05 by igrr