espressif/esp-idf #3321: [ESP HTTP CLIENT]: When re-init the esp http client, the client always down with "TRANS_SSL: mbedtls_net_connect returned -52"

elitezhu, 2019-04-17:
I used the ESP http client to send data to MindSphere which is Siemens's cloud platform.  As I check the http client API, it doesn't provide a http client context clearing interface in case that
esp_http_client_perform() fails. So I have no choice but use the API of esp_http_client_cleanup() to free http client totally and re-init a new http client. The code shows as below:
E_MCL_ERROR_CODE http_client_send(http_client_t *http_client, http_request_t *http_request, http_client_send_callback_info_t *callback_info, http_response_t **http_response)
{
    E_MCL_ERROR_CODE    code        = MCL_FAIL;
    esp_err_t           err         = ESP_OK;
    mcl_size_t          i           = 0;
    

    string_t *          reqheader   = 0;
    char *              key         = 0;
    char *              value       = 0;
    char *              colon       = 0;

    if (http_client->pHandle == NULL) {
        return MCL_FAIL;
    }
  
    for (i = 0; i < http_request->header->index; i++) { 
        reqheader    = http_request->header->items[i].string;

        colon  = strchr(reqheader->buffer, ':');

        *colon = 0;
        key    = reqheader->buffer;
        value  = colon+2;
        
        // MCL_INFO("%s: %s", key, value);
        err = esp_http_client_set_header(http_client->pHandle, key, value);
        
        *colon = ':';

        if (err != ESP_OK) {
            MCL_INFO("HTTP client header setting... FAIL: %s", esp_err_to_name(err));
            break;
        }
    }

    if (err == ESP_OK) {
        err = esp_http_client_set_url(http_client->pHandle, http_request->uri->buffer);

        if (err != ESP_OK){
            MCL_INFO("HTTP client URL setting... FAIL: %s", esp_err_to_name(err));
        }
    } 

   
   
    if (err == ESP_OK) {
        switch(http_request->method) {
            case MCL_HTTP_GET:
                err = esp_http_client_set_method(http_client->pHandle, HTTP_METHOD_GET);
                break;
                    
            case MCL_HTTP_PUT:
                err = esp_http_client_set_method(http_client->pHandle, HTTP_METHOD_PUT);
                break;
                    
            case MCL_HTTP_POST:
                if (callback_info == NULL) {
                    err = esp_http_client_set_method(http_client->pHandle, HTTP_METHOD_POST);
                }
                break;
                    
            default:
                err = ESP_FAIL;
                break;
        }

        if (err != ESP_OK) {
[... 152 lines cut ...]

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

elitezhu, 2019-04-18:
Hi, ESP32 development team,
ESP http client state machine management process seems not very good for exception case, like expiring the timeout. when it happens the http client just return a error code but not handle the state machine code. So when I re-send the last failed-sending telegram, the http client is in a wrong state.

jitin17, 2019-04-25:
`esp-http-client` doesn't support retry logic. It is the responsibility of the application to handle redirection.
From the snippet you have posted here,  it is difficult to figure out where exactly is the issue. Would you mind sharing the entire application?

elitezhu, 2019-04-28:
@jitin17 , Can you share your e-mail address?

jitin17, 2019-05-02:
@elitezhu Can you share it at this address: `jitin at espressif dot com`?

shubhamkulkarni97, 2020-01-02:
@MaoniinoaM Is your issue resolved?

Alvin1Zhang, 2020-03-05:
@MaoniinoaM Thanks for reporting. Will close the ticket due to short of feedback. Feel free to reopen with more details. Thanks.

closed 2020-03-05 by Alvin1Zhang