espressif/esp-idf #3323: Multiple ESP32 stations disconnect simultaneously.

arossoinsite, 2019-04-17:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v3.1.3-2-g0d825c6e7
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: external 5V

## Problem Description

The company that I work for, has some boards based on the ESP32-WROOM-32U module and installed 10 of them for a customer a couple of months ago. The boards are distributed throughout the customer house which has also various APs, some of them are close to the boards, so WiFi signal strength is not a problem. The boards connected to the network without problem when installed, but every two weeks approximately, all of them or the vast majority, lose connection to the AP (almost simultaneously) and do not connect again until they are power-cycled or the routers restarted, the last one is the main workaround we've used to reconnect the boards. We have installed a kind of _probe_ board from which we capture logs and we have seen too many `wifi: bcn_timout,ap_probe_send_start` logs on it. Also, we've looked at routers logs and see that after the boards lose connection, the router receives the DHCPDISCOVER message from boards and sends DHCPOFFER, but never receives DHCPREQUEST.

### Expected Behavior
The ESP should send DHCPREQUEST message to go ahead with DHCP negotiation and recover connectivity.
### Actual Behavior
ESP sends DHCPDISCOVER message but after router sends DHCPOFFER, there is no response from ESP, so it nevers get IP again.

### Some relevants code fragments
This is the function used to handle wifi events, it just call some previous registered functions, I list only the relevant events:
```cpp
esp_err_t wifi_event_handler(void *ctx, system_event_t *event)
{
   switch (event->event_id)
    {
    /* Some other events here... */
    case SYSTEM_EVENT_STA_LOST_IP:
        ESP_LOGD("WIFI_EVENT", "EVENT SYSTEM_EVENT_STA_LOST_IP\n");
        wifi_state.got_ip = 0;
        if (!helpers_isNull(wifi_callbacks[wifi_callback_sta_lost_ip]))
            wifi_callbacks[wifi_callback_sta_lost_ip](ctx, event);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        ESP_LOGD("WIFI_EVENT", "EVENT SYSTEM_EVENT_STA_DISCONNECTED\n");
        ESP_LOGW("WIFI EVENT", "Wi-Fi Reason: %s (%u)", wifi_reason2str(event->event_info.disconnected.reason), event->event_info.disconnected.reason);
        if (!helpers_isNull(wifi_callbacks[wifi_callback_sta_disconnected]))
            wifi_callbacks[wifi_callback_sta_disconnected](ctx, event);
        break;
    return WIFI_OK;
}
```
The relevant callbacks:
```cpp
#define NETWORK_MANAGER_MAX_DISCONNECTS_BEFORE_AP   20
#define WIFI_WAIT_IP_SECS      (30)

void _network_manager_callback_sta_disconnected(void *ctx, system_event_t *event)
{
   ++network_manager_state.wifi_connection_attempts;
    ESP_LOGI(TAG, "Connection attempt: %d", network_manager_state.wifi_connection_attempts);
    
    int f_ret = wifi_get_wifi_mode();
    network_manager_state.is_wifi_connected = 0;
    HELPERS_FREE_PTR(network_manager_state.wifi_ip);
    if (network_manager_state.wifi_connection_attempts >= NETWORK_MANAGER_MAX_DISCONNECTS_BEFORE_AP)
    {
      if (f_ret != wifi_MODE_ACCESS_POINT)
      {
        network_manager_stablish_ap(1);
      }
      
      return;
    }
    
    if (wifi_MODE_STATION == wifi_get_wifi_mode()) {
      wifi_connect();
    }
}

void _network_manager_callback_sta_lost_ip(void *ctx, system_event_t *event)
{
 ESP_LOGW(TAG,"sta_lost_ip");
  network_manager_state.is_wifi_connected = 0;
  HELPERS_FREE_PTR(network_manager_state.wifi_ip);
  
  if (network_manager_state.wait_ip_task == NULL) {
    xTaskCreatePinnedToCore(
      _network_manager_wait_ip,
      "_network_manager_wait_ip",
      4096,
      NULL,
      2,
      &network_manager_state.wait_ip_task,
      tskNO_AFFINITY);
  }
}

void _network_manager_callback_sta_got_ip(void *ctx, system_event_t *event)
{
  ESP_LOGD(TAG,"_network_manager_callback_sta_got_ip");
  ESP_LOGI(TAG, "Got IP!");
  if (wifi_get_ip(&network_manager_state.wifi_ip, event) == 0)
  {
    if (network_manager_state.wifi_ip != NULL)
    {
      network_manager_state.is_wifi_connected = 1;
    }
  }
  
  xSemaphoreGive(network_manager_state.wait_ip_semphr);
  
  network_manager_state.wifi_connection_attempts = 0;
  if (network_manager_state.reconnectHandle != NULL)
  {
    vTaskDelete(network_manager_state.reconnectHandle);
    network_manager_state.reconnect_task = 0;
    network_manager_state.reconnectHandle = NULL;
  }
  
  if (network_manager_state.wait_ip_task != NULL) {
    vTaskDelete(network_manager_state.wait_ip_task);
    network_manager_state.wait_ip_task = NULL;
  }
}

static void _network_manager_wait_ip(void *pvParameters)
{
  BaseType_t ret;
  unsigned int retry_cnt = 3;
  
  xSemaphoreTake(network_manager_state.wait_ip_semphr, portMAX_DELAY);
  
  for (;;) {
    ESP_LOGI(TAG, "Waiting IP...");
    
    ret = xSemaphoreTake(
        network_manager_state.wait_ip_semphr,
        (1000 * WIFI_WAIT_IP_SECS) / portTICK_PERIOD_MS);
    
    if (pdTRUE != ret) {
      if (0U == retry_cnt) {
        ESP_LOGW(TAG, "Wait IP timeout! Restarting...");
        esp_restart();
      }
      else {
        ESP_LOGW(TAG, "Wait IP timeout!");
        wifi_shutdown();
        vTaskDelay(10);
        network_manager_try_join_wifi();
        retry_cnt--;
      }
    }
    
    xSemaphoreGive(network_manager_state.wait_ip_semphr);
    network_manager_state.wait_ip_task = NULL;
    vTaskDelete(NULL);
  }
}
```
Some relevant functions:
```cpp
int network_manager_try_join_wifi()
{
  char *ssid = NULL;
  char *password = NULL;
  /* Load stored credentials from NVS. */
  int fun_return = network_manager_get_wifi_credentials(&ssid, &password);
  if (NETWORK_MANAGER_OK != fun_return){
    goto exit;
  }
  network_manager_connect_to_wifi(ssid, password, false);
  goto exit;

  exit:
    HELPERS_FREE_PTR(ssid);
    HELPERS_FREE_PTR(password);
    return fun_return;
}

```

## Debug Logs
Logs from the _probe_ board:
```
I (408911917) wifi: bcn_timout,ap_probe_send_start
I (408920927) wifi: bcn_timout,ap_probe_send_start
I (408929937) wifi: bcn_timout,ap_probe_send_start
I (417880847) wifi: bcn_timout,ap_probe_send_start
I (418607887) wifi: bcn_timout,ap_probe_send_start
I (418610387) wifi: ap_probe_send over, resett wifi status to disassoc
I (418610387) wifi: state: run -> init (100)
I (418610387) wifi: pm stop, total sleep time: 0 us / 351265433152 us

I (418610397) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
D (418610397) WIFI_EVENT: EVENT SYSTEM_EVENT_STA_DISCONNECTED

W (418610407) WIFI EVENT: Wi-Fi Reason: BEACON_TIMEOUT (200)
I (418610407) wifi: flush txq
I (418610417) wifi: stop sw txq
I (418610417) wifi: lmac stop hw txq
```
```
I (133447) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (134007) wifi: state: init -> auth (b0)
I (134017) wifi: state: auth -> assoc (0)
I (134017) wifi: state: assoc -> run (10)
I (134027) wifi: connected with STELLA, channel 11
I (134027) wifi: pm start, type: 0

I (162177) NETWORK_MANAGER: trying to reconnect to network
D (162177) Wi-Fi: wifi_disconnect()
I (162177) wifi: state: run -> init (0)
I (162177) wifi: pm stop, total sleep time: 0 us / 28144400 us

I (162177) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
D (162187) WIFI_EVENT: esp_wifi_disconnect:0
D (162187) WIFI_EVENT: EVENT SYSTEM_EVENT_STA_DISCONNECTED
W (162197) WIFI EVENT: Wi-Fi Reason: ASSOC_LEAVE (8)
I (162217) wifi: flush txq
I (162227) wifi: stop sw txq
I (162237) wifi: lmac stop hw txq
```
Logs from the router:
[image]

## sdkconfig file
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3091661/sdkconfig.txt)

negativekelvin, 2019-04-18:
You may want to try changing this to no

`CONFIG_LWIP_DHCP_DOES_ARP_CHECK=y`

burkulesomesh43, 2019-05-06:
Any Update?? 
I am also facing same problem. As I kept my ESP32 live for long time, it disconnects and not getting connected to the configured SSID and PASSWORD again till switch it OFF and ON again.
please tell me if issue is solved or any other solution.

arossoinsite, 2019-05-15:
We have not found a solution yet, but it seems that setting a static IP for ESP's makes the connection fails less frequently.

zhangyanjiaoesp, 2019-06-24:
@arossoinsite Does this problem still exist?

arossoinsite, 2019-06-25:
@zhangyanjiaoesp We discovered that the reason for the simultaneous disconnection was that occasionally someone turn off and on the mains supply of the house, but we never knew why the boards didn't obtain IP again. After #3474 solution, the boards have been online for several days without problems.

closed 2019-06-25 by arossoinsite