espressif/esp-idf #3325: Unable to set I2S samplerate under 2441Hz - ADC

rob-bits, 2019-04-18:
## Environment
- Development Kit:      ESP32-Gateway
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v3.3-release

## Problem Description

I want to use the I2S with internal or external 16bit ADC with samplerate of 1000 Hz. But based on the i2s driver the samplerate cannot be set under 2441 Hz.

In the line 317 from the i2s.c module this can be found:

```cpp
    double clkmdiv = (double)I2S_BASE_CLK / (rate * factor);

    if (clkmdiv > 256) {
        ESP_LOGE(I2S_TAG, "clkmdiv is too large\r\n");
        return ESP_ERR_INVALID_ARG;
    }

```

The I2S_BASE_CLK is 160MHz, the factor is 256 and based on the equation 2441Hz will be the lowest which can be set.

Is this a harware issue? Or is it required a different configuration if we want to reach slower speeds?

### Code to reproduce this issue

```cpp
  if(type == I2S_EXTERNAL_ADC)
  {
    i2s_config_flags = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_RX);

    i2s_in_pin_config = 
    {
      bck_io_num: ADC_SCLK,
      ws_io_num: ADC_CONVST,
      data_out_num: -1, //Not used
      data_in_num: ADC_DOUT
    };
  }
  else
  {
    i2s_config_flags = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN);
  }

i2s_config_t i2s_config =
  {
    //.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_TX | I2S_MODE_ADC_BUILT_IN),  // I2S receive mode with ADC
    .mode = i2s_config_flags,  // I2S receive mode with ADC
    .sample_rate = 1000,                                                          // sample rate
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,                                 // 16 bit I2S
    .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,                                 // only the right channel+
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,                                  //
    //.communication_format = I2S_COMM_FORMAT_I2S,                                  // I2S format
    .intr_alloc_flags = 0,                                                        // none
    //.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,                                                        // none
    .dma_buf_count = 16,                                                           // number of DMA buffers, 128 is the max
    .dma_buf_len = U8_NUM_SAMPLES,                                                   // number of samples, 1024 is the max
    //.use_apll = 0,                                                                // no Audio PLL
    .use_apll = true,                                                                // no Audio PLL
    .tx_desc_auto_clear = 1,     //I2S auto clear tx descriptor if there is underflow condition (helps in avoiding noise in case of data unavailability)
    .fixed_mclk = 1000
  };

  // install I2S 0 driver, using event queue
  i2s_driver_install(I2S_NUM_0, &i2s_config, 1, &i2s_event_queue);

```

## Debug Logs

```
E (2765) I2S: clkmdiv is too large

E (2766) I2S: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2s.c:844 (i2s_set_sample_rates):bits_per_sample not set
Guru Meditation Error: Core 1 panic'ed (IntegerDivideByZero). Exception was unhandled.
```

koobest, 2019-04-18:
Hi, 
Maybe your `fixed_mclk` is too small. can you try to set `fixed_mclk = 40000000` ?

rob-bits, 2019-05-23:
Any suggestion? Or would somebody link a documentation for APLL configuration?

koobest, 2019-05-23:
Hi,@rob-bits
Do you need to change the sampling rate frequently?

rob-bits, 2019-05-23:
Hi, @koobest 
Frequently means each time when user want to change the acqusition speed. In the end software the user will have control on the samplerate. And probably the samplerate will be set to one value for minutes/hours but if the end-user want to measure at different sampling rate then he will change it.

koobest, 2019-05-23:
Hi, @rob-bits
Can you accept the following solutions?  when the sampling rate is greater than 2441, use `i2s_set_sample_rates` to set the clock. when the sample rate is less than 2441,  you can configure the clock as follows(change the value of  clkm_div_num and tx_bck_div_num):
```
    i2s_stop(0);
   rtc_clk_apll_enable(1, 32, 32, 2, 31);
    I2S0.clkm_conf.clkm_div_num = 40;
    I2S0.clkm_conf.clka_en = 1;  //enable APLL
    I2S0.clkm_conf.clkm_div_b = 0;
    I2S0.clkm_conf.clkm_div_a = 1;
    I2S0.sample_rate_conf.tx_bck_div_num = 40;
    I2S0.sample_rate_conf.rx_bck_div_num = 40;
    i2s_start(0);
```

rob-bits, 2019-05-24:
Hi, @koobest,

Thanks for the suggestion! I will try it out and get back to you.
If it is working then I will close this issue.

rob-bits, 2019-06-19:
Hi @koobest,

I am trying your suggestion and seems it work. But I would like to control the requested frequency. Do you know how can I calculate the output frequency by equation? I tried the equation from the ESP32 technical reference manual from the chapter "12.3 The Clock of I2S Module" but I got different values. So the measured acqusition frequency and the calculated frequency are not equal.

Thank you in advance!

L-KAYA, 2022-02-24:
Hi @koobest, it's true that a low sample rate is not supported before `release/v4.4`, now the algorithm of APLL has been optimized, you can refer to [clk_ctrl_os.c on master](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/clk_ctrl_os.c) for the latest APLL algorithm and line 929/980 in [i2s.c](https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c) for the calculation of ADC clock.

closed 2022-05-05 by Alvin1Zhang