espressif/esp-idf #3327: Put the nvs partition behind the ota_1 partition, which is the location between 0x320000 and 0x3fffff, the system crashes.

GitOfLance, 2019-04-18:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version : v4.0-dev-207-g76da10a66
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
I want to increase the size of the `nvs` partition, but if I don't move the location of the nvs partition, I can only use 20KB when using the `ota` partition and the `factory` partition. I want to try 104KB of space, so I put nvs at the end of the partition table, ready to use the space between `0x320000` and `0x3FFFFF`, but even if I only allocate 4kb size here will make the system crash.
I found an image as shown below:
[image]

- According to the picture, can the last 896KB Flash space be allocated as a partition?
- What is this 896KB Flash used for?
- How many bytes can the nvs partition allocate? Is there a size limit?
- Are all the relevant information in [Partition Tables](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html#partition-tables)?Is there any other information?

### Steps to repropduce
- step 1: Modify the partition table file as follows
```
# Name,   Type, SubType, Offset,   Size
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
# nvs_1,      data, nvs,     ,         0x4000
otadata,  data, ota,     ,         0x2000
phy_init, data, phy,     ,         0x1000
factory,  0,    0,       ,         1M
coredump, data, coredump,,         64K
ota_0,    0,    ota_0,   ,         1M
ota_1,    0,    ota_1,   ,         1M
nvs_2,    data, nvs,	 ,         0x1000
```
- step 2: `make menuconfig`, select this partition.csv
- step 3: `make flash monitor`

### Debug Logs
System crash information：
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6372
load:0x40078000,len:11308
load:0x40080400,len:6680
entry 0x40080760
I (29) boot: ESP-IDF v4.0-dev-207-g76da10a66-dirty 2nd stage bootloader
I (29) boot: compile time 01:10:23
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (66) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot:  3 coredump         Unknown data     01 03 00110000 00010000
I (89) boot:  4 ota_0            OTA app          00 10 00120000 00100000
I (96) boot:  5 ota_1            OTA app          00 11 00220000 00100000
I (103) boot:  6 nvs_2            WiFi data        01 02 00320000 00001000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x194b8 (103608) map
I (165) esp_image: segment 1: paddr=0x000294e0 vaddr=0x3ffb0000 size=0x02ff0 ( 12272) load
I (170) esp_image: segment 2: paddr=0x0002c4d8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/lance/Workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (172) esp_image: segment 3: paddr=0x0002c8e0 vaddr=0x40080400 size=0x03730 ( 14128) load
I (187) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x68d34 (429364) map
0x400d0018: _stext at ??:?

I (340) esp_image: segment 5: paddr=0x00098d54 vaddr=0x40083b30 size=0x0cc40 ( 52288) load
0x40083b30: spi_flash_mmap_pages at /home/lance/Workspace/esp/esp-idf/components/spi_flash/flash_mmap.c:320

I (372) boot: Loaded app from partition at offset 0x10000
I (372) boot: Disabling RNG early entropy source...
I (373) cpu_start: Pro cpu up.
I (376) cpu_start: Application information:
I (381) cpu_start: Project name:     esp32_ilock10
I (386) cpu_start: App version:      1
I (391) cpu_start: Compile time:     Apr 18 2019 01:10:27
I (397) cpu_start: ELF file SHA256:  ab9f73ab4505ac2c...
I (403) cpu_start: ESP-IDF:          v4.0-dev-207-g76da10a66-dirty
I (410) cpu_start: Starting app cpu, entry point is 0x400810d8
0x400810d8: call_start_cpu1 at /home/lance/Workspace/esp/esp-idf/components/esp32/cpu_start.c:267

I (402) cpu_start: App cpu up.
I (420) heap_init: Initializing. RAM available for dynamic allocation:
I (427) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (433) heap_init: At 3FFB8DF0 len 00027210 (156 KiB): DRAM
I (440) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (446) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (452) heap_init: At 40090770 len 0000F890 (62 KiB): IRAM
I (458) cpu_start: Pro cpu start user code
I (141) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4008800c
0x4008800c: _esp_error_check_failed at /home/lance/Workspace/esp/esp-idf/components/esp32/panic.c:720

file: "/home/lance/Workspace/esp/ilock10_esp32/main/main.c" line 354
func: init_nvs_flash_ilock
expression: ret

ELF file SHA256: ab9f73ab4505ac2c30aa4febce5308184382556d133e64d226fa545431fb3c24

Backtrace: 0x40087ba0:0x3ffbab50 0x4008800f:0x3ffbab70 0x400d3051:0x3ffbab90 0x400d305b:0x3ffbabb0 0x400d107e:0x3ffbabe0 0x4008cc79:0x3ffbac00
0x40087ba0: invoke_abort at /home/lance/Workspace/esp/esp-idf/components/esp32/panic.c:715

0x4008800f: _esp_error_check_failed at /home/lance/Workspace/esp/esp-idf/components/esp32/panic.c:721

0x400d3051: init_nvs_flash_ilock at /home/lance/Workspace/esp/ilock10_esp32/main/main.c:354 (discriminator 1)

0x400d305b: app_main at /home/lance/Workspace/esp/ilock10_esp32/main/main.c:370

0x400d107e: main_task at /home/lance/Workspace/esp/esp-idf/components/esp32/cpu_start.c:530

0x4008cc79: vPortTaskWrapper at /home/lance/Workspace/esp/esp-idf/components/freertos/port.c:403
```
Use the `make partition_table` to get the partition table information：
```
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
otadata,data,ota,0x9000,8K,
phy_init,data,phy,0xb000,4K,
factory,app,factory,0x10000,1M,
coredump,data,coredump,0x110000,64K,
ota_0,app,ota_0,0x120000,1M,
ota_1,app,ota_1,0x220000,1M,
nvs_2,data,nvs,0x320000,4K,
*******************************************************************************
```

### Steps to repropduce
Initialize the code of the nvs partition:
```
/* nvs flash init */
void init_nvs_flash_ilock()
{
	// Initialize NVS
    esp_err_t ret = nvs_flash_init();
	
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );
}
```

igrr, 2019-04-18:
@GitOfLance According to the documentation of [`nvs_flash_init`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#_CPPv414nvs_flash_initv), it expects the partition name to be `nvs`. Error `ESP_ERR_NOT_FOUND` in your log indicates that `nvs` partition is not being found. In your partition table the name is `nvs_2`. If you wish to use a different name, you can `esp_err_tnvs_flash_init_partition` function instead.

NVS partition size should be at least 12kB. The upper limit is determined by the amount of heap memory available (NVS consumes some amount of heap memory for caching, and this is proportional to the partition size). Also the larger the partition is, the longer `nvs_flash_init` or `nvs_flash_init_partition` will take. As a rule of thumb, if you find that you need NVS partition larger than 64kB, you should consider using SPIFFS or FATFS for your storage needs instead. 

>According to the picture, can the last 896KB Flash space be allocated as a partition?
>What is this 896KB Flash used for?

This space is not used for anything in the default partition table. You can define custom partitions there.

GitOfLance, 2019-04-18:
@igrr Thank you very much for your solution to the problem.  I ignored the function of `nvs_flash_init` and thank you for your advice on using SPIFFS or FATFS to store large amounts of data. I will try the method you suggested.
Thanks.

Alvin1Zhang, 2019-07-19:
@GitOfLance Thanks for reporting the issue, would you help share if any updates for the issue? Thanks.

GitOfLance, 2019-07-20:
@Alvin1Zhang The problem is solved.According to the suggestion，use `nvs` to store key-value pairs and use the file system to store large amounts of data.There is nothing to add. You can close this issue now.
Thank you！

Alvin1Zhang, 2020-04-29:
@Alvin1Zhang Thanks for feedback.

closed 2020-04-29 by Alvin1Zhang

tingjhenjiang, 2024-08-11:
> @GitOfLance According to the documentation of [`nvs_flash_init`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html#_CPPv414nvs_flash_initv), it expects the partition name to be `nvs`. Error `ESP_ERR_NOT_FOUND` in your log indicates that `nvs` partition is not being found. In your partition table the name is `nvs_2`. If you wish to use a different name, you can `nvs_flash_init_partition` function instead.
> 
> NVS partition size should be at least 12kB. The upper limit is determined by the amount of heap memory available (NVS consumes some amount of heap memory for caching, and this is proportional to the partition size). Also the larger the partition is, the longer `nvs_flash_init` or `nvs_flash_init_partition` will take. As a rule of thumb, if you find that you need NVS partition larger than 64kB, you should consider using SPIFFS or FATFS for your storage needs instead.
> 
> > According to the picture, can the last 896KB Flash space be allocated as a partition?
> > What is this 896KB Flash used for?
> 
> This space is not used for anything in the default partition table. You can define custom partitions there.

@igrr 
In using the last 896KB Flash space,
after I use customized partition table, an error occurred indicating that "Partitions tables occupies 5.7MB of flash (5963775 bytes) which does not fit in configured flash size 4MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu." But after I change the flash size in menuconfig and build and flash the firmware again, ESP32 has another error occurred, indicating that "spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed." 

What should I do? Thanks.

igrr, 2024-08-11:
If you have a 4 MB flash chip, you have to arrange the partitions on such a way that the total size doesn't exceed 4 MB.