espressif/esp-idf #3330: Bootloader sdk v3.2 cannot load app -> Fatal exception (28): LoadProhibited

farizadp, 2019-04-19:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-PICO-D4
- IDF version v3.2
- Build System:         Make
- Compiler version crosstool-ng-1.22.0-80-g6c4433a
- Operating System:     Linux
- Power Supply:         USB

## Problem Description

I migrated sdk from a255622 to v3.2 (286202c) and our bootloader build is not able to load application anymore. All build succeeds

The only thing I change in out bootloader source code is ```esp_image_load(ESP_IMAGE_LOAD, ``` to ```esp_image_verify(ESP_IMAGE_VERIFY, ```

I disable all secure boot

### Expected Behavior
As seen below ```app: ``` text which is part of the app is shown. This is part of our application.
```
D (281) boot: Firmware update CRC @ 0x13fffc: 39a8d98a
I (287) boot: Calculating CRC32 for partition at 0x10000
D (293) bootloader_flash: mmu set block paddr=0x00010000 (was 0x00130000)
D (327) bootloader_flash: mmu set block paddr=0x00020000 (was 0x00010000)
D (354) bootloader_flash: mmu set block paddr=0x00030000 (was 0x00020000)
D (382) bootloader_flash: mmu set block paddr=0x00040000 (was 0x00030000)
D (409) bootloader_flash: mmu set block paddr=0x00050000 (was 0x00040000)
D (437) bootloader_flash: mmu set block paddr=0x00060000 (was 0x00050000)
D (464) bootloader_flash: mmu set block paddr=0x00070000 (was 0x00060000)
D (492) bootloader_flash: mmu set block paddr=0x00080000 (was 0x00070000)
D (520) bootloader_flash: mmu set block paddr=0x00090000 (was 0x00080000)
D (547) bootloader_flash: mmu set block paddr=0x000a0000 (was 0x00090000)
D (575) bootloader_flash: mmu set block paddr=0x000b0000 (was 0x000a0000)
D (602) bootloader_flash: mmu set block paddr=0x000c0000 (was 0x000b0000)
D (630) bootloader_flash: mmu set block paddr=0x000d0000 (was 0x000c0000)
D (657) bootloader_flash: mmu set block paddr=0x000e0000 (was 0x000d0000)
D (685) bootloader_flash: mmu set block paddr=0x000f0000 (was 0x000e0000)
D (712) bootloader_flash: mmu set block paddr=0x00100000 (was 0x000f0000)
D (740) bootloader_flash: mmu set block paddr=0x00110000 (was 0x00100000)
D (767) bootloader_flash: mmu set block paddr=0x00120000 (was 0x00110000)
D (795) bootloader_flash: mmu set block paddr=0x00130000 (was 0x00120000)
D (822) boot: Calculated CRC @ offset 0x10000: 0x39a8d98a
I (822) boot: Firmware CRC32 check OK (0x39a8d98a)
D (823) boot: Firmware verified succesfully
I (826) boot: Booting firmware at 0x00010000
I (831) boot: Disabling RNG early entropy source...
D (837) esp_image: reading image header @ 0x10000
D (842) bootloader_flash: mmu set block paddr=0x00010000 (was 0x00130000)
D (848) esp_image: image header: 0xe9 0x06 0x02 0x02 40080ef8
I (854) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x38e1c (232988) map
D (863) bootloader_flash: mmu set paddr=00010000 count=4
D (950) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
I (950) esp_image: segment 1: paddr=0x00048e44 vaddr=0x3ffb0000 size=0x035c8 ( 13768) load
D (955) bootloader_flash: mmu set paddr=00040000 count=1
D (965) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
I (967) esp_image: segment 2: paddr=0x0004c414 vaddr=0x40080000 size=0x00400 (  1024) load
D (976) bootloader_flash: mmu set paddr=00040000 count=1
D (982) bootloader_flash: mmu set block paddr=0x00040000 (was 0xffffffff)
I (988) esp_image: segment 3: paddr=0x0004c81c vaddr=0x40080400 size=0x037f4 ( 14324) load
D (997) bootloader_flash: mmu set paddr=00040000 count=2
D (1008) bootloader_flash: mmu set block paddr=0x00050000 (was 0xffffffff)
I (1009) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xa7e84 (687748) map
D (1018) bootloader_flash: mmu set paddr=00050000 count=11
D (1265) bootloader_flash: mmu set block paddr=0x000f0000 (was 0xffffffff)
I (1265) esp_image: segment 5: paddr=0x000f7ea4 vaddr=0x40083bf4 size=0x0cb20 ( 52000) load
D (1269) bootloader_flash: mmu set paddr=000f0000 count=2
D (1296) bootloader_flash: mmu set block paddr=0x00100000 (was 0xffffffff)
D (1296) esp_image: Calculated hash: 1a8e86c1
D (1296) bootloader_flash: mmu set paddr=00100000 count=1
[... 74 lines cut ...]
D (70863) boot: calling set_cache_and_start_app
D (70870) boot: configure drom and irom and start
D (70879) boot: start: 0x40080ef8
I (70886) Fatal exception (28): LoadProhibited
epc1=0x40008044, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000008e6, depc=0x00000000
```

Behavior changes I notice are below.
* printout between ```bootloader_flash: mmu set block paddr``` texts is much slower
* printout between ```esp_image: segment ``` is also much slower
* ```esp_image: Calculated hash: ``` prints the entire hash (early sdk prints only the first 8)

### Partition table
```
#Name,          Type,   SubType,    Offset,     Size
# not used,     ,       ,           0x0,        0x1000
#stage2_bl,     app,    bl,         0x1000,     0x7000
#partitions,    data,   pt,         0x8000,     0x1000
nvs,            data,   nvs,        0x9000,     0x4000
otadata,        data,   ota,        0xD000,     0x3000
firmware_0,     app,    ota_0,      0x10000,    0x130000
firmware_1,     app,    ota_1,      0x140000,   0x130000
confprim,       data,   nvs,        0x3A0000,   0xC000
confback,       data,   nvs,        0x3AC000,   0xC000
```

igrr, 2019-04-19:
Hi @farizadp, could you please attach the .map and .elf files of the bootloader?

farizadp, 2019-04-19:
Sure, here you go
[bootloader.zip](https://github.com/espressif/esp-idf/files/3098181/bootloader.zip)

farizadp, 2019-04-19:
@igrr : What kind of things I can expect to find in the map file related to the issue?

negativekelvin, 2019-04-20:
Did it really take 70 seconds or is the clock messed up?

igrr, 2019-04-22:
@farizadp My plan was to:

1) check that there are no functions called from the image loading stage which are placed into .text segment, they should be in .iram_loader.text segment. See https://github.com/espressif/esp-idf/blob/a20d02b7f196c407bc9f39b781e31a0a4f665968/components/bootloader/subproject/main/esp32.bootloader.ld#L35-L69

2) convert the bootloader to a .bin file, flash it onto a board, and then debug it over JTAG

With regards to step 1, i have noticed three functions which normally seem to be not present in IDF bootloader: `verify_firmware`, `get_signed_crc32`, `calculate_crc32`. Are you using a customized bootloader? If yes, please note that the bootloader component has been reorganized between IDF v3.1 and v3.2. I have also noticed that the functions in the ELF file resemble ESP-IDF v3.1 bootloader, not v3.2 bootloader.
For correct operation, you may need to redo your modifications on top of a stock v3.2 bootloader. The positive thing about this is that it should now be much easier to modify the bootloader, as only the top level logic is now located in the bootloader component. The support functions are in `bootloader_support`, so it should now be easier to update IDF to new version without losing any changes.

With regards to step 2, unfortunately I couldn't reproduce the issue, as the firmware is not getting loaded seemingly due to the added check.
```
I (31) boot: Starting PT ESP32 2nd stage bootloader
...
E (270) boot: 'entry_point' equals nullptr, error initializing boot partition
E (280) boot: No valid firmware found, aborting boot and exiting.
user code done
```

farizadp, 2019-04-30:
@igrr : Thanks, and yes those functions are created by us specifically for our application. And yes, we thought that updating only the SDK and replacing deprecated functions used by our implementation would be sufficient.

In that case, we'll try to re-implement our bootloader functionality on top of v3.2. Will this always be the case for future SDK releases?

projectgus, 2019-06-04:
Hi @farizadp,

Sorry for the delay between responses. Did you get the bootloader working correctly in v3.2?

> In that case, we'll try to re-implement our bootloader functionality on top of v3.2. Will this always be the case for future SDK releases?

This is our mistake that we didn't highlight the potential for breaking changes due to the bootloader refactoring in v3.2. Sorry about that.

The idea behind the bootloader refactoring is to present a clearer public API via the bootloader_support component for all major functions of the bootloader, so the ESP-IDF default "bootloader" subproject itself becomes a minimal implementation that mostly calls bootloader_support functions.

This means that a custom bootloader can be created without needing to directly modify much ESP-IDF internal code. A custom bootloader that only uses the bootloader_support API, plus custom code that calls bootloader_support API functions, should not break during future minor IDF updates. If the bootloader_support API behaviour has breaking changes as part of a major IDF update then we'll flag it as a breaking change in the release notes. ([IDF version policy is here](https://docs.espressif.com/projects/esp-idf/en/latest/versions.html)).

If a custom bootloader involves changing or replacing the implementation of the bootloader_support component, rather than only calling the APIs in bootloader_support, then it'll need to be checked carefully during each update as the implementation may have changed between releases.

If you have particular requirements regarding bootloader customization that are awkward even after the refactoring in v3.2, then please open a new issue with as many requirements details as you can provide, and we'll try to find ways to support it in the future.

Alvin1Zhang, 2019-10-11:
@farizadp Thanks for reporting, would you please help share if you get the bootloader working correctly in v3.2? Thanks.

farizadp, 2019-10-14:
We put this issue on hold for now. I close this thread for now (may be reopened later on)

closed 2019-10-14 by farizadp