espressif/esp-idf #3336: how use AUTOBAUD in esp32?

xiaosanli, 2019-04-20:
i find it's have union of auto_baud in the struct uart_dev_t, but don't find use it any where in esp-idf. how use the auto_baud function? who can tell me, thanks!!

negativekelvin, 2019-04-20:
It is in the trm, but basically it just measures bit length and then then you can calculate the baud.

koobest, 2019-04-22:
Hi,xiaosanli

You should enable the register `UART_AUTOBAUD_EN`, and than, the sender needs to send a binary sequence such as `01010101`. The receiver will detect the shortest pulse. The minimum duration is stored in `UART_LOWPULSE_REG` and `UART_HIGHPULSE_REG`. 

baud_rate = 80000000/min(UART_LOWPULSE_REG, UART_HIGHPULSE_REG);

thanks !!

Alvin1Zhang, 2019-05-20:
@xiaosanli Hi, thanks for reporting the issue. Do you have some time to try the approaches suggested by @koobest ? Would you help share if any updates for the issue? Thanks.

Alvin1Zhang, 2019-05-27:
@xiaosanli Thanks for reporting the issue, feel free to reopen the issue if you are still unclear on how to use autobaud. Thanks.

closed 2019-05-27 by Alvin1Zhang