espressif/esp-idf #3337: spi_slave_initialize produces "spi: not using iomux pins" error when using correct VSPI IOMUX pins

tntclaus, 2019-04-20:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version: ESP-IDF v4.0-dev-402-ga20d02b7f
- Build System:         CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-73-ge28a011) 5.2.0
- Operating System:     macOS
- Power Supply:         USB

## Problem Description

SPI slave/master initialisation with explicit IOMUX flag causes `spi: not using iomux pins` error. 

Docs explicitly state that IOMUX is available for HSPI and VSPI pins, but the latest esp-idf v4.0-dev produces errors for both master and slave initialisation on VSPI pin group.

Changing pins to HSPI group works as expected without any errors using the same sample code.

### Expected Behavior

`spi_slave_initialize(VSPI_HOST, &buscfg, 1)` and `spi_bus_initialize(VSPI_HOST, &buscfg, 1);` return `ESP_OK`.

### Actual Behavior

Instead they both produce:

```
E (282) spi: not using iomux pins
E (282) spi: spicommon_bus_initialize_io(231): not all required capabilities satisfied.
```

### Steps to repropduce

1. Run code below on any wroom device.

### Code to reproduce this issue

This is slightly modified spi_slave example. You can change `spi_slave_initialize(VSPI_HOST, &buscfg, 1)` to `spi_bus_initialize(VSPI_HOST, &buscfg, 1);` for checking SPI master.

```cpp
#include <stdio.h>
#include <stdint.h>
#include <stddef.h>
#include <string.h>
#include <driver/gpio.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"
#include "lwip/igmp.h"

#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "soc/rtc_cntl_reg.h"
#include "rom/cache.h"
#include "driver/spi_slave.h"
#include "esp_log.h"
#include "esp_spi_flash.h"

//Pins in use. The SPI Master can use the GPIO mux, so feel free to change these if needed.
#define GPIO_MOSI 23
#define GPIO_MISO 19
#define GPIO_SCLK 18
#define GPIO_CS 5

//Main application
void app_main() {
    int n = 0;
    esp_err_t ret;

    //Configuration for the SPI bus
    spi_bus_config_t buscfg = {
            .mosi_io_num=GPIO_MOSI,
            .miso_io_num=GPIO_MISO,
            .sclk_io_num=GPIO_SCLK,
            .flags = SPICOMMON_BUSFLAG_NATIVE_PINS
    };

    //Configuration for the SPI slave interface
    spi_slave_interface_config_t slvcfg = {
            .mode=0,
            .spics_io_num=GPIO_CS,
            .queue_size=1,
            .flags=0
            //.post_trans_cb=my_post_trans_cb
    };

    //Enable pull-ups on SPI lines so we don't detect rogue pulses when no master is connected.
    gpio_set_pull_mode(GPIO_MISO, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_SCLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_CS, GPIO_PULLUP_ONLY);

//    gpio_iomux_in(GPIO_MOSI, VSPID_OUT_IDX);
//    gpio_iomux_out(GPIO_MISO, VSPIQ_IN_IDX, false);
//    gpio_iomux_in(GPIO_SCLK, VSPICLK_IN_IDX);
//    gpio_iomux_in(GPIO_CS, VSPICS0_IN_IDX);

    //Initialize SPI slave interface
    ret = spi_slave_initialize(VSPI_HOST, &buscfg, &slvcfg, 1);
    assert(ret == ESP_OK);
}
```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6328
load:0x40078000,len:11308
load:0x40080400,len:6700
entry 0x40080764
I (28) boot: ESP-IDF v4.0-dev-402-ga20d02b7f 2nd stage bootloader
I (28) boot: compile time 02:50:17
I (28) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0b7c4 ( 47044) map
I (109) esp_image: segment 1: paddr=0x0001b7ec vaddr=0x3ffb0000 size=0x01eb0 (  7856) load
I (113) esp_image: segment 2: paddr=0x0001d6a4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/klaus/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (117) esp_image: segment 3: paddr=0x0001daac vaddr=0x40080400 size=0x02564 (  9572) load
I (129) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x14254 ( 82516) map
0x400d0018: _flash_cache_start at ??:?

I (163) esp_image: segment 5: paddr=0x00034274 vaddr=0x40082964 size=0x05cc4 ( 23748) load
0x40082964: esp_timer_impl_get_time at /Users/klaus/esp/esp-idf/components/esp32/esp_timer_esp32.c:340

I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
I (179) cpu_start: Pro cpu up.
I (182) cpu_start: Application information:
I (187) cpu_start: Project name:     spi-slave
I (192) cpu_start: App version:      1
I (197) cpu_start: Compile time:     Apr 20 2019 02:49:50
I (203) cpu_start: ELF file SHA256:  bb9824e0ae4b1733...
I (209) cpu_start: ESP-IDF:          v4.0-dev-402-ga20d02b7f
I (215) cpu_start: Starting app cpu, entry point is 0x40080e88
0x40080e88: call_start_cpu1 at /Users/klaus/esp/esp-idf/components/esp32/cpu_start.c:267

I (0) cpu_start: App cpu up.
I (226) heap_init: Initializing. RAM available for dynamic allocation:
I (233) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (239) heap_init: At 3FFB2F00 len 0002D100 (180 KiB): DRAM
I (245) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (251) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (258) heap_init: At 40088628 len 000179D8 (94 KiB): IRAM
I (264) cpu_start: Pro cpu start user code
I (282) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (284) spi: not using iomux pins
E (284) spi: spicommon_bus_initialize_io(231): not all required capabilities satisfied.
```

negativekelvin, 2019-04-20:
Try
```
 //Configuration for the SPI bus
    spi_bus_config_t buscfg = {
            .mosi_io_num=GPIO_MOSI,
            .miso_io_num=GPIO_MISO,
            .sclk_io_num=GPIO_SCLK,
            .quadwp_io_num=-1,
            .quadhd_io_num=-1
            .flags = SPICOMMON_BUSFLAG_NATIVE_PINS
    };
```

tntclaus, 2019-04-20:
@negativekelvin well, thank you. That was so stupid from my side not to try setting quadwp/quadhd to -1 when I saw it in SPI master example.

closed 2019-04-20 by tntclaus