espressif/esp-idf #3338: [vfs_fat.c] is fat_drive too short ?

tobozo, 2019-04-20:
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version : [v4.1]
- Module or chip used:  [ESP32-WROVER-I]
- IDF version a20d02b 
- Build System:         [Make|CMake]
- Operating System:     [Windows|Linux]
- Power Supply:         [USB|]

## Problem Description

SD Cards with drive label longer than 8 chars may cause unexpected result.

https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/vfs_fat.c#L28

    char fat_drive[8]; /* FAT drive name */

as [highilighted by @phillowcompiler]((https://twitter.com/phillowcompiler/status/1118154909445005313)), shouldn't it be 11 instead of 8 ?

negativekelvin, 2019-04-20:
I don't think so because the volume name is not actually used for that.

https://github.com/espressif/esp-idf/blob/a20d02b7f196c407bc9f39b781e31a0a4f665968/components/fatfs/src/vfs_fat_sdmmc.c#L97-L100

tobozo, 2019-04-21:
so this means even if my fat drive name has more than 8 chars it won't affect anything ?

[image]

phillowcompiler, 2019-04-21:
Ooooh,I seee!

Thanks your research & information!(^^)
I'll resarch more,too...

igrr, 2022-03-14:
As correctly pointed out by negativekelvin, fat_drive only stores the volume name used by FatFS library (such as `0:`), not the volume label.

closed 2022-03-14 by igrr