espressif/esp-idf #3339: [fatfs/src/ff.c] getFileInfo 8.3 test false positives

tobozo, 2019-04-20:
- Development Kit:      [ESP32-Wrover-Kit]
- Kit version : [v4.1]
- Module or chip used:  [ESP32-WROVER-I]
- IDF version a20d02b 
- Build System:         [Make|CMake]
- Operating System:     [Windows|Linux]
- Power Supply:         [USB|]

## Problem Description

https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ff.c#L2732

    if (IsUpper(wc) && (dp->dir[DIR_NTres] & ((si >= 9) ? NS_EXT : NS_BODY))) wc += 0x20;

This test fails in two situations: 
  - short filenames ( < 9 chars ) 
  - number-only filenames ( i.e. `2048.bin` returns `2048.BIN` )

also highlighted by @phillowcompiler and @lovyan03

 [source](https://twitter.com/phillowcompiler/status/1118144919087001600)
 [file name examples](https://twitter.com/lovyan03/status/1118011630770462725)

phillowcompiler, 2019-04-21:
+ number-only or [UPPER CASE only ] filenames(ie. ABCD.exe retruns ABCD.EXE )

phillowcompiler, 2019-04-21:
For example,
I modified as follows.

> int dotflg = 0;
> 			for (si = di = 0; fno->altname[si]; si++, di++) {	/* Copy altname[] to fname[] with case information */
> 				wc = (WCHAR)fno->altname[si];
> 				if (IsUpper(wc) && (dp->dir[DIR_NTres] & ((dotflg) ? NS_EXT : NS_BODY))) wc += 0x20;
> 				fno->fname[di] = (TCHAR)wc;
>                 if(wc=='.'){dotflg = 1;}
> 			}
> 
> 

so that, I confirmed correct file names by ESP-IDF code at like "while(){readdir()}".

igrr, 2019-07-08:
This has been fixed in 8f6606141a673887385dcdfa7012d60aaa66d148 by updating Fatfs to r0.13c patch 4.

closed 2019-07-08 by igrr